PixlOne/logiops #353: GUI for Linux

Eko450-eng, 2023-03-16:
Hey guys I have started using logiops yesterday and was searching for a gui to work with it, I couldn't find a gui for it so I decided to make one myself and have made really great progress so far basically finished a working prototype.

Since I have little knowledge about the logiops in generell other than what I have been using it for I'd really appreciate some ideas and maybe some people to help me out with just getting the general functionality and maybe some ui ideas if anyone is interested in having a look at it and help out let me know I'll post a video with what I have gotten so far later today.

(Sorry if this is the wrong issue format or wrong way to post it)

Eko450-eng, 2023-03-17:
Update: https://github.com/Eko450-eng/LogiOpsGui/releases/tag/Linux

aleksejalex, 2023-03-25:
thanks, will definitely try it! ...But why only Master 3S is listed as supported? Is it really or it's just the only mouse tested with this GUI so far?

Eko450-eng, 2023-04-19:
It is the only one I have tested so far.