PixlOne/logiops #355: Gestures button not being recognised!

NavTech270, 2023-03-18:
Even though my house definitely has a gesture button, anything I map the gesture button to do never works. I would like to open activities overview but this doesn't seem to be working.

`Mouse: M720 Triathlon Multi-Device Mouse`

My Config:

```
devices: ({
  name: "M720 Triathlon Multi-Device Mouse";

  hiresscroll: {
    hires: true;
    invert: false;
    target: false;
  };

  dpi: 1500; // max=4000

  buttons: (
    // Forward button
    {
      cid: 0x56;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_BACK" ];
            }
          }
	);
      };
    },

  // Back button
    {
      cid: 0x53;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_FORWARD" ];
            }
          }
	);
      };
    },

    // Gesture button (hold and move)
    {
      cid: 0xc3;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA" ]; // open activities overview
            }
          }
	);
      };
    },

    // Middle button
    {
      cid: 0x52;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "KeyPress";
              keys: ["KEY_RIGHTMETA" ];
            }
          },

          {
            direction: "Up";
            mode: "OnRelease";
            action = {
              type: "ChangeDPI";
              inc: 1000,
            }
          },

          {
            direction: "Down";
            mode: "OnRelease";
            action = {
              type: "ChangeDPI";
              inc: -1000,
            }
          }
        );
      };
    }
  );
});
```
[image]

NavTech270, 2023-03-18:
What is also the config in order to close tab via clicking scroll wheel button? @PixlOne

theonejj, 2023-04-01:
this config is working for me on the m720
https://raw.githubusercontent.com/menuRivera/logiops/master/logid.cfg

NavTech270, 2023-10-04:
Thank you! This helped a lot with helping me know what CID I had to use for the gesture button and syntax for gestures. @theonejj

closed 2023-10-04 by NavTech270