PixlOne/logiops #361: No DJ reports

fpafumi, 2023-04-29:
Hi, I have this error when I run `sudo logid -v`
```
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
[DEBUG] Unsupported device /dev/hidraw10 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw11 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[INFO] Detected receiver at /dev/hidraw12
[WARN] Error adding device /dev/hidraw12: No DJ reports
[DEBUG] Unsupported device /dev/hidraw8 ignored
[DEBUG] Unsupported device /dev/hidraw9 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
```
My configuration in _/etc/logid.cfg_ is:
```
devices: (
{
    name: "MX Master 3S";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 4000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_UP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_DOWN"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "CycleDPI";
                            dpis: [400, 600, 800, 1000, 1200, 1400, 1600];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type = "ToggleSmartshift";
                        }
                    },
                    {
                        direction: "None"
                        mode: "NoPress"
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type: "Keypress";
                keys: ["KEY_A"];
            };
        }
    );
}
);

```

The mouse is **mx master 3s**.
Thanks all for your support.

PixlOne, 2023-04-30:
The MX Master 3S was not supported previously due to changes in the Logi Bolt receiver. I added support recently alongside many other changes, please try the feature/ipcgull branch (see #362).

fpafumi, 2023-04-30:
great job! it works correctly but during compiling the code I has some warning.
thanks.

closed 2023-05-03 by PixlOne