PixlOne/logiops #366: Gestures with direction: "None" not working after upgrading to 0.3.x

amusingaxl, 2023-05-03:
I just upgraded to v0.3.0 and my previous config is now broken.

I noticed that none of the gestures with `direction: "None"` work anymore. All others continue to work just fine.

Here's my config file for reference:

```cfg
// File location: /etc/logid.cfg
devices: (
  {
    name: "Wireless Mouse MX Master 3",

    smartshift: {
      on: true,
      threshold: 100,
      default_threshold: 100,
    },

    hiresscroll: {
      hires: true,
      invert: false,
      target: false,
      up: {
        mode: "Axis",
        axis: "REL_WHEEL_HI_RES",
        axis_multiplier: 1.0,
      },
      down: {
        mode: "Axis",
        axis: "REL_WHEEL_HI_RES",
        axis_multiplier: -1.0,
      },
    },

    dpi: 2000,

    thumbwheel: {
      divert: true,
      invert: false,
      left: {
        mode: "Axis",
        axis: "REL_HWHEEL",
        axis_multiplier: -0.5,
      },
      right: {
        mode: "Axis",
        axis: "REL_HWHEEL",
        axis_multiplier: 0.5,
      },
    },

    buttons: (
      # Forward button
      {
        cid: 0x56,
        action: {
          type: "Gestures",
          gestures: (
            {
              direction: "None", # NOT WORKING!
              mode: "OnRelease",
              action: {
                type: "Keypress",
                # keys: [ "KEY_LEFTALT", "KEY_RIGHT" ],
                keys: [ "KEY_FORWARD" ],
              }
            },

            {
              direction: "Right",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTMETA", "KEY_LEFTSHIFT", "KEY_RIGHT" ], // Move to the monitor on the right
              }
            },

            {
              direction: "Left",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTMETA", "KEY_LEFTSHIFT", "KEY_LEFT" ], // Move to the monitor on the left
              }
            },

            {
              direction: "Down",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTMETA", "KEY_D" ], // show desktop
              }
            },

            {
              direction: "Up",
              threshold: 500,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTALT", "KEY_F4" ], // close window
              }
            },
          ),
        },
      },
      # Back button
      {
        cid: 0x53,
        action: {
          type: "Gestures",
          gestures: (
            {
              direction: "None", # NOT WORKING!
              mode: "OnRelease",
              action: {
                type: "Keypress",
                # keys: [ "KEY_LEFTALT", "KEY_LEFT" ],
                keys: [ "KEY_BACK" ],
              }
            },
            # {
            #   direction: "Up",
            #   threshold: 100,
            #   mode: "OnRelease",
            #   action: {
            #     type: "Keypress",
            #     keys: [ "KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_UP" ], // open workspaces overview
            #   }
            # },
            {
              direction: "Up",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTMETA", "KEY_LEFTALT", "KEY_A" ], // toggle show window in all workspaces
              }
            },
            {
              direction: "Down",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_DOWN" ], // open activities overview
              }
            },
            {
              direction: "Right",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_RIGHT" ],
              }
            },

            {
              direction: "Left",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_LEFT" ],
              }
            },
          ),
        },
      },
      {
        # Gesture button (hold and move)
        cid: 0xc3,
        action: {
          type: "Gestures",
          gestures: (
            {
              direction: "None", # NOT WORKING!
              threshold: 1,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTCTRL", "KEY_0" ], // Reset zoom level
              }
            },
	        {
              direction: "Up",
              threshold: 100,
              mode: "onRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTALT", "KEY_F10" ], // toggle maximize window
              }
	        },
	        {
              direction: "Down",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTALT", "KEY_F6" ], // minimize window
              }
            },
            {
              direction: "Right",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTMETA", "KEY_RIGHT" ], // snap window to right
              }
            },
            {
              direction: "Left",
              threshold: 100,
              mode: "OnRelease",
              action: {
                type: "Keypress",
                keys: [ "KEY_LEFTMETA", "KEY_LEFT" ], // snap window to left
              }
	        },
          ),
        },
      },
      # Top button
      {
        cid: 0xc4,
        action: {
          type: "Gestures",
          gestures: (
            {
              direction: "None", # NOT WORKING!
              mode: "OnRelease",
              action: {
                type: "ToggleSmartShift",
              }
            },

            {
              direction: "Up",
              threshold: 500,
              mode: "OnRelease",
              action: {
                type: "ChangeDPI",
                inc: 1000,
              }
            },
            {
              direction: "Down",
              threshold: 500,
              mode: "OnRelease",
              action: {
                type: "ChangeDPI",
                inc: -1000,
              }
            }
          ),
        },
      },
    ),
  },
  {
    name: "MX Keys Wireless Keyboard",
    buttons: (
      {
        cid: 0x00c7,
        action: {
          type: "keypress",
          keys: [ "KEY_BRIGHTNESSDOWN" ],
        }
      },

      {
        cid: 0x00c8,
        action: {
          type: "keypress",
          keys: [ "KEY_BRIGHTNESSUP" ],
        }
      }
    ),
  },
)
```

PixlOne, 2023-05-03:
Just tested this out, reproduced it, and fixed in a recent commit.

closed 2023-05-03 by PixlOne

amusingaxl, 2023-05-03:
Any ETA for this make into a release? I'm currently not installing it from souce.

commit 5e436a2b (2023-05-03): Fix none gesture not working

Also fixes gestures on MX Master 3S by ignoring the first movement
event. Fixes #366 and #339.