PixlOne/logiops #367: Make Command Failing

calcelli, 2023-05-03:
I am receiving the errors
`../logiops/src/logid/./config/types.h:354:50: error: class template argument deduction failed:
  354 |         struct config_io<map<K, V, KeyName, Cmp, Alloc>> {
`

`logiops/src/logid/./config/types.h:354:50: error: no matching function for call to ‘string_literal(string_literal<...auto...>)’`

`logiops/src/logid/./config/map.h:29:19: note: candidate: ‘template<long unsigned int N> string_literal(const char (&)[N])-> logid::config::string_literal<N>’
   29 |         constexpr string_literal(const char (& str)[N]) {
`

Amongst many others when I am following the compile instructions in the ReadME.

I am sorry if this is a duplicate or the wrong place to ask. I tried looking through open issues and could not find it

PixlOne, 2023-05-03:
So I've just tried building on Ubuntu 20.04, turns out it still uses GCC 9 which doesn't have full C++20 support. I found some other compile issues on GCC 10 which I'll fix soon. In any case, I'll update dependencies and figure out how to put a better warning for this.

calcelli, 2023-05-03:
Okay Cool, So it should work on 22?

PixlOne, 2023-05-03:
Should now work on 22 or by upgrading g++ to version 10, I'll try fixing the errors to get it working on g++ 9

PR #369 (2023-05-03): Fix c++2a compatibility
Adds gcc 9 support and fixes builds on Ubuntu 20.04, fixes #367
closed 2023-05-03 by PixlOne