PixlOne/logiops #370: always scroll up with version 0.3

LeoDT, 2023-05-05:
it works well in 0.2, but after upgrading to 0.3, the scrolling is always up, can not scroll down.

config file:

```
devices: (
{
    name: "Wireless Mobile Mouse MX Anywhere 2S";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: true;
        up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 2;
        };
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -2;
        };
    };
    dpi: 1500;
    buttons: (
        {
            cid: 0x5b;
            action = { type: "Keypress"; keys: ["BTN_SIDE"]; };
        },
        {
            cid: 0x5d;
            action = { type: "Keypress"; keys: ["BTN_EXTRA"]; };
        },
        {
            cid: 0x53;
            action = { type: "Keypress"; keys: ["BTN_MIDDLE"]; };
        }
    );
}
);
```

bittamer, 2023-05-06:
I'm having the same problem on Arch Linux. Here's the config file:

```devices: ({
  name: "MX Anywhere 3";

  smartshift: {
    on: true;
    threshold: 15;
  };

  hiresscroll: {
    hires: true;
    invert: true;
    target: true;
       up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 1;
        },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -1;
        },
  };

  dpi: 1000; // max=4000

    buttons: (
        {
            cid: 0x52;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Left";
                        mode: "OnInterval";
			interval: 10;
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_VOLUMEDOWN"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnInterval";
			interval: 10;
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_VOLUMEUP"];
                        };
                    },
                    {
                        direction: "None"
                        mode: "OnRelease";
                        action =
                        {
			    type: "Keypress";
			    keys: ["BTN_MIDDLE"];
                        }
                    }
                );
            };
        },
        {
            cid: 0x53;
            action =
            {
                type: "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_I"];
            };
	},
        {
            cid: 0x56;
            action =
            {
                type: "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_DOT"];
            };
	}
    );
}
);
```

Using `axis: "REL_WHEEL"` instead of `axis: "REL_WHEEL_HI_RES"` works, though. But I'd rather use hi res scrolling.

tpaniaki, 2023-05-06:
Scrolling up/down not working altogether on my system (Arch/Gnome):

```
devices:	(	{
    name: "Wireless Mouse MX Master 3";
    
    smartshift: 	{on: false;
        			threshold: 30;};
    
    hiresscroll:	{hires: false;
        			invert: false;
        			target: false;};
    dpi: 1000;

   
    buttons: 	(     
       				{cid: 0xc3; 
       					action = 	{type: "Gestures"; gestures:	(	{direction: "Right"; mode: "OnRelease"; action = {type: "Keypress"; keys: ["KEY_LEFTALT", "KEY_LEFTSHIFT", "KEY_J"];}},
       																	{direction: "Left"; mode: "OnRelease"; action = {type: "Keypress"; keys: ["KEY_LEFTALT", "KEY_R"];}},
       																	{direction: "Down"; mode: "OnRelease"; action = {type: "Keypress"; keys: ["KEY_LEFTALT", "KEY_Y"];}},
       																	{direction: "Up"; mode: "OnRelease"; action = {type: "Keypress"; keys: ["KEY_LEFTALT", "KEY_Q"];}},
       																	{direction: "None"; mode: "OnRelease"; action = {type: "Keypress"; keys: ["KEY_LEFTALT", "KEY_G"]};}
       					
       					
       																)
       								}
       															
       				;}	
				)	
				}	
			);

```

Stegorex, 2023-05-06:
+1. Arch (Manjaro), KDE Plasma. Hires scroling does not work at all, without hires, both wheel directions are scrolling up

impankratov, 2023-05-06:
MX Master 2S:
Changed `target: true` to `target: false` like in [example config](https://github.com/PixlOne/logiops/blob/main/logid.example.cfg):
```
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };

```
And scrolling started working again

Stegorex, 2023-05-06:
> MX Master 2S: Changed `target: true` to `target: false` like in [example config](https://github.com/PixlOne/logiops/blob/main/logid.example.cfg):
> 
> ```

> ```
> 
> And scrolling started working again

Thank you for the workaround, it re-enables scrolling for me, too. Only disadvantage is, that the defined multipliers of course  have no effect when disabling the target, which makes scrolling less efficient than I'd like it to be, but this is still much better than not being able to scroll at all :-)

S-trace, 2023-05-25:
Hi all.

I was affected too, but have resolved this problem.
Just use the following config section as an example:
```
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: true;
        up: {
            mode: "Axis";
            axis: "REL_WHEEL";
            axis_multiplier: 1.0;
        };
        down: {
            mode: "Axis";
            axis: "REL_WHEEL";
            axis_multiplier: -1.0;
        };
    };
```

Important notes: 
1. axis_multiplier should be float, not int. You can't use `axis_multiplier: -1;`, you must use `axis_multiplier: -1.0;` or you will have always scroll up problem.
2. axis should be "REL_WHEEL", you can't use "REL_WHEEL_HI_RES" or you will have no scroll at all (possibly kernel bug - I can see REL_WHEEL_HI_RES in `evtest`, but the scrolling does not work).

tpaniaki, 2023-06-07:
Nice workaround. However, now I get incremental scrolling even if the wheel is set loose (MX Master 3). I guess the "REL_WHEEL_HI_RES" option was intended to avoid this?

Sdghasemi, 2024-02-07:
I had the same issue on my MX Master 3S, migrating back to v0.2.4 was not an option for me because of its various bugs. I used this PR to fix the problem on v3.3.0. Now everything works fine thanks to the author.
https://github.com/PixlOne/logiops/pull/413