PixlOne/logiops #371: Fatal error starting logid

superstator, 2023-05-08:
I am trying to get `logid` up and running on an Ubuntu 22.04 system with an MX Mechanical + MX Master 3S. I was able to pull down the latest (`be840b333`) and build / install with no issue, but the process core dumps on startup:

```
$ sudo systemctl start logid
$ sudo systemctl status logid
× logid.service - Logitech Configuration Daemon
     Loaded: loaded (/lib/systemd/system/logid.service; disabled; vendor preset: enabled)
     Active: failed (Result: core-dump) since Mon 2023-05-08 14:46:36 PDT; 14s ago
    Process: 8686 ExecStart=/usr/bin/logid (code=dumped, signal=ABRT)
   Main PID: 8686 (code=dumped, signal=ABRT)
        CPU: 6ms

May 08 14:46:35 onibaba systemd[1]: Started Logitech Configuration Daemon.
May 08 14:46:35 onibaba logid[8686]: [INFO] Config file does not exist, using empty config.
May 08 14:46:35 onibaba logid[8686]: [INFO] Detected receiver at /dev/hidraw7
May 08 14:46:36 onibaba logid[8686]: [ERROR] Orphan Receiver, missing DeviceManager
May 08 14:46:36 onibaba logid[8686]: [ERROR] Fatal error, file a bug report. Program will now exit.
May 08 14:46:36 onibaba logid[8686]: terminate called without an active exception
May 08 14:46:36 onibaba systemd[1]: logid.service: Main process exited, code=dumped, status=6/ABRT
May 08 14:46:36 onibaba systemd[1]: logid.service: Failed with result 'core-dump'.
```

So here I am filing a bug report. FWIW I have seen it enumerate the MX Mechanical in that list at least once, but never the MX Master. Anything worth trying in the config file?

PixlOne, 2023-05-11:
So what likely happened is that the GDBus server died suddenly and the program was set to exit. This seems to have happened at the same time the receiver was trying to add a device. The program should exit more gracefully in that case though but I guess that''s a different issue.

Is the service running as root? Is DBus running? Does `/usr/share/dbus-1/system.d/pizza.pixl.LogiOps.conf` exist?

superstator, 2023-05-11:
Yes, `logid.service` is running as root (I just used the default `make install` setup). DBus is running, and I can call it from the cmdline (e.g. `gdbus call --session --dest 'org.gnome.Shell' --object-path '/org/gnome/Shell' --method 'org.gnome.Shell.ShowOSD' "{'icon': <'audio-volume-muted'>, 'label': <'Test'>, 'level': <10>}"`. `systemctl` says it has been running continuously since the machine booted, with no suspicious log messages. `/usr/share/dbus-1/system.d/pizza.pixl.LogiOps.conf` does not exist.

PixlOne, 2023-05-11:
Strange, that file should be installed when running `make install`. What's the output of CMake? Does it say `dbus system policy will be installed at ...`?

loganmill, 2023-05-15:
I experienced the identical issue as _superstator_: I'm also running ubuntu 22.04.  

"make install" did not install pizza.pixl.LogiOps.conf, and the message "dbus system policy will be installed at ..." was not generated by make install.  Seems this is  because the test

   if(DBUS_FOUND)

at line  114 of CMakeLists.txt evaluated to false, though I've no reason to believe dbus was not installed on my system.   However, installing libdbus-glib-1-dev fixed it for me.  Perhaps consider adding this to the DEBIAN/UBUNTU dependencies in README.md?

PixlOne, 2023-05-15:
I think I'll instead remove this check, DBus is required anyways so it should fail before that point. I really don't know why I put that there, I guess I was planning on allowing users to remove DBus support if they want to, but I suppose that's not worth the hassle right now.

Thanks for debugging that @loganmill

PR #378 (2023-05-16): Remove check if DBus is found
Broken on Ubuntu and leads to DBus policy not being installed, DBus is required anyways right now so remove this check. Fixes #371.
commit b4a9d446 (2023-05-16): Remove check if DBus is found

Broken on Ubuntu and leads to DBus policy not being installed, DBus
is required anyways right now so remove this check. Fixes #371.
closed 2023-05-16 by PixlOne