PixlOne/logiops #374: Logid v0.3.2 hangs after initialization

S-trace, 2023-05-09:
Hi.
I have two devices (K850 Performance Wireless Keyboard and Wireless Mouse MX Master 3) connected over bluetooth.
I'm having issues with the latest version of logid (v0.3.2+ be840b33). The initialization of devices goes fine, but the program hangs shortly after that.
After startup it prints out the following:
```
$ sudo ./logid -v
[DEBUG] Unsupported device /dev/hidraw0 ignored
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw1:255
[WARN] ScrollDown is not a direction
[WARN] ScrollUp is not a direction
[WARN] ScrollDown is not a direction
[WARN] ScrollUp is not a direction
[DEBUG] /dev/hidraw1:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
[INFO] Device found: K850 Performance Wireless Keyboard on /dev/hidraw2:255
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x01 | YES     | YES     |            | 
[DEBUG] 0x02 | YES     | YES     |            | 
[DEBUG] 0x03 | YES     | YES     |            | 
[DEBUG] 0x04 | YES     | YES     |            | 
[DEBUG] 0x05 | YES     | YES     |            | 
[DEBUG] 0x06 | YES     | YES     |            | 
[DEBUG] 0x0a | YES     | YES     |            | 
[DEBUG] 0x34 |         |         |            | 
[DEBUG] 0xbd | YES     | YES     |            | 
[DEBUG] 0xbf | YES     | YES     |            | 
[DEBUG] 0xc0 | YES     | YES     |            | 
[DEBUG] 0xc1 | YES     | YES     |            | 
[DEBUG] 0xc2 | YES     | YES     |            | 
[DEBUG] 0xd1 | YES     |         |            | 
[DEBUG] 0xd2 | YES     |         |            | 
[DEBUG] 0xd3 | YES     |         |            | 
[DEBUG] 0xd4 | YES     | YES     |            | 
[DEBUG] 0xd5 | YES     | YES     |            | 
[DEBUG] 0xd6 | YES     | YES     |            | 
[DEBUG] 0xdd | YES     | YES     |            | 
[DEBUG] 0xde |         | YES     |            | 
```
And then hangs. 

I can reboot my mouse or keyboard, and nothing is printed to the log after startup, and gestures are not processed (sometimes I can do one or two gestures before initialization completes, but logid stops processing gestures after an initialization (when I'm doing gesture - the cursor freezes, nothing happens).

My good old logid `v0.2.2-79-g7383c6f` build works fine in this situation:
```
sudo logid -v
[DEBUG] Unsupported device /dev/hidraw0 ignored
[INFO] Device K850 Performance Wireless Keyboard not configured, using default config.
[INFO] Device found: K850 Performance Wireless Keyboard on /dev/hidraw2:255
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x01 | YES     | YES     |            | 
[DEBUG] 0x02 | YES     | YES     |            | 
[DEBUG] 0x03 | YES     | YES     |            | 
[DEBUG] 0x04 | YES     | YES     |            | 
[DEBUG] 0x05 | YES     | YES     |            | 
[DEBUG] 0x06 | YES     | YES     |            | 
[DEBUG] 0x0a | YES     | YES     |            | 
[DEBUG] 0x34 |         |         |            | 
[DEBUG] 0xbd | YES     | YES     |            | 
[DEBUG] 0xbf | YES     | YES     |            | 
[DEBUG] 0xc0 | YES     | YES     |            | 
[DEBUG] 0xc1 | YES     | YES     |            | 
[DEBUG] 0xc2 | YES     | YES     |            | 
[DEBUG] 0xd1 | YES     |         |            | 
[DEBUG] 0xd2 | YES     |         |            | 
[DEBUG] 0xd3 | YES     |         |            | 
[DEBUG] 0xd4 | YES     | YES     |            | 
[DEBUG] 0xd5 | YES     | YES     |            | 
[DEBUG] 0xd6 | YES     | YES     |            | 
[DEBUG] 0xdd | YES     | YES     |            | 
[DEBUG] 0xde |         | YES     |            | 
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw1:255
[WARN] Line 42: target is true but no up action was set, using default
[WARN] Line 42: target is true but no down action was set, using default
[DEBUG] /dev/hidraw1:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
#### logid completely started ####
[WARN] Error ignored on detached thread/task: _readReport read failed: Input/output error
[INFO] Device on /dev/hidraw1 disconnected
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw1:255
[WARN] Line 42: target is true but no up action was set, using default
[WARN] Line 42: target is true but no down action was set, using default
[DEBUG] /dev/hidraw1:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
[WARN] Error ignored on detached thread/task: _readReport read failed: Input/output error
[INFO] Device on /dev/hidraw2 disconnected
[INFO] Device K850 Performance Wireless Keyboard not configured, using default config.
[INFO] Device found: K850 Performance Wireless Keyboard on /dev/hidraw2:255
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x01 | YES     | YES     |            | 
[DEBUG] 0x02 | YES     | YES     |            | 
[DEBUG] 0x03 | YES     | YES     |            | 
[DEBUG] 0x04 | YES     | YES     |            | 
[DEBUG] 0x05 | YES     | YES     |            | 
[DEBUG] 0x06 | YES     | YES     |            | 
[DEBUG] 0x0a | YES     | YES     |            | 
[DEBUG] 0x34 |         |         |            | 
[DEBUG] 0xbd | YES     | YES     |            | 
[DEBUG] 0xbf | YES     | YES     |            | 
[DEBUG] 0xc0 | YES     | YES     |            | 
[DEBUG] 0xc1 | YES     | YES     |            | 
[DEBUG] 0xc2 | YES     | YES     |            | 
[DEBUG] 0xd1 | YES     |         |            | 
[DEBUG] 0xd2 | YES     |         |            | 
[DEBUG] 0xd3 | YES     |         |            | 
[DEBUG] 0xd4 | YES     | YES     |            | 
[DEBUG] 0xd5 | YES     | YES     |            | 
[DEBUG] 0xd6 | YES     | YES     |            | 
[DEBUG] 0xdd | YES     | YES     |            | 
[DEBUG] 0xde |         | YES     |            | 
```

PixlOne, 2023-05-11:
If you run `sudo logid -v RAWREPORT`, do you see any output after the point where it normally freezes? Do you also see any raw reports output from devices other than `/dev/hidraw1` or `/dev/hidraw2`?

S-trace, 2023-05-12:
No, I don't see any output after freeze.

Also, sometimes logid fails to initialize my keyboard and prints `[DEBUG] Failed to add device /dev/hidraw1 on try 1, backing off for 500ms` right before freeze (and nothing is printed after this line, even if I reboot the mouse or the keyboard).

I don't see raw reports output from other devices (I tried to swipe the touchpad and tap the built-in laptop keyboard - nothing was printed).

Here is a full log from `sudo logid -v RAWREPORT` (I have updated logid to the latest commit 30ade71e):
[logid_rawreport_30ade71e.txt](https://github.com/PixlOne/logiops/files/11457174/logid_rawreport_30ade71e.txt)

PixlOne, 2023-05-13:
What input devices are connected? Can you run `ls /dev/hidraw*`?

wmbruckner, 2023-05-14:
Hi,
I have the same problem since an update was done via opensuse tumbleweed. Here's the rawreport with the listing of hidraw. Lat command was aborted while trying to run logid manually (hangs).
wb@linux:~> sudo logid -v RAWREPORT
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[RAWREPORT] /dev/hidraw6 OUT: 10 ff 00 12 00 00 00 
[RAWREPORT] /dev/hidraw6 IN:  11 01 00 12 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[DEBUG] Ignoring virtual node on /dev/hidraw6
[RAWREPORT] /dev/hidraw4 OUT: 10 ff 00 12 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  10 ff 8f 00 12 01 00 
[RAWREPORT] /dev/hidraw4 OUT: 10 ff 00 12 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  10 ff 8f 00 12 01 00 
[INFO] Detected receiver at /dev/hidraw4
[RAWREPORT] /dev/hidraw4 OUT: 10 ff 00 12 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  10 ff 8f 00 12 01 00 
[RAWREPORT] /dev/hidraw4 OUT: 10 ff 00 12 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  10 ff 8f 00 12 01 00 
[RAWREPORT] /dev/hidraw4 OUT: 10 ff 81 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  10 ff 81 00 00 09 00 
[RAWREPORT] /dev/hidraw4 OUT: 10 ff 80 00 10 09 00 
[RAWREPORT] /dev/hidraw4 IN:  10 ff 80 00 00 00 00 
[RAWREPORT] /dev/hidraw4 OUT: 10 ff 80 02 02 00 00 
[RAWREPORT] /dev/hidraw4 IN:  10 01 41 04 b2 69 40 
[RAWREPORT] /dev/hidraw4 OUT: 10 01 00 12 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  10 ff 80 02 00 00 00 
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[RAWREPORT] /dev/hidraw4 IN:  11 01 00 01 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 OUT: 10 01 00 12 00 00 68 
[RAWREPORT] /dev/hidraw4 IN:  11 01 00 12 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 00 01 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 00 12 04 05 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 00 01 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 07 01 32 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 07 01 32 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 07 01 32 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 07 11 04 04 c0 28 0a 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 07 11 04 04 c0 28 0a 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 07 11 04 04 c0 28 0a 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw4 IN:  11 01 00 01 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[DEBUG] Failed to add device /dev/hidraw4:1 on try 1, backing off for 250ms
^C
wb@linux:~> ls /dev/hidraw*
/dev/hidraw0  /dev/hidraw1  /dev/hidraw2  /dev/hidraw3  /dev/hidraw4  /dev/hidraw5  /dev/hidraw6
wb@linux:~> sudo logid
[INFO] Detected receiver at /dev/hidraw4
^C
logid version is called "null" by logid itself, installer zypper lists version as logiops-1683772204.30ade71-5.1.x86_64

Before, logiops ran just fine

PixlOne, 2023-05-14:
Could you try out the fix-iomon-locks branch?

S-trace, 2023-05-18:
> Could you try out the fix-iomon-locks branch?

I tried fix-iomon-locks branch at commit f85cd5b and master branch at commit 64962a8 - no luck, logid hangs with all those builds. 

> What input devices are connected? Can you run `ls /dev/hidraw*`?

```

PixlOne, 2023-05-20:
@S-trace I pushed another change on there, can you see if that fixes it?

S-trace, 2023-05-21:
@PixlOne I have checked cb7a2da and 5767aac and now logid just exits after initialization without any messages.

PixlOne, 2023-05-22:
Possibly related to #371? Does `/usr/share/dbus-1/system.d/pizza.pixl.LogiOps.conf` exist?

S-trace, 2023-05-22:
> Possibly related to #371? Does `/usr/share/dbus-1/system.d/pizza.pixl.LogiOps.conf` exist?

No, /usr/share/dbus-1/system.d/pizza.pixl.LogiOps.conf does not exist in my system.

PixlOne, 2023-05-23:
That's likely the issue. I need to figure out a better way to get logid to output this error. Are you installing the new versions or just building them and running them? I should have resolved this issue in #371.

PixlOne, 2023-05-24:
You need to install them so that DBus configuration files are installed as well, it also must be run as root if you're not doing that.

S-trace, 2023-05-25:
> You need to install them so that DBus configuration files are installed as well, it also must be run as root if you're not doing that.

Thanks, I have executed `make install` and now logid v0.3.2 (upstream/fix-iomon-locks commit @cb7a2dad) works.