PixlOne/logiops #376: Logid fails to start with systemctl, runs fine when executed manually

tejus, 2023-05-12:
OS: Ubuntu 22.04 with KDE (Plasma X11 mode)

Logid works well when run from a terminal window as `sudo logid`. However enabling on systemctl shows this status:

```
× logid.service - Logitech Configuration Daemon
     Loaded: loaded (/lib/systemd/system/logid.service; enabled; vendor preset: enabled)
     Active: failed (Result: core-dump) since Fri 2023-05-12 14:13:23 IST; 5min ago
    Process: 10234 ExecStart=/usr/bin/logid (code=dumped, signal=ABRT)
   Main PID: 10234 (code=dumped, signal=ABRT)
        CPU: 8ms

May 12 14:13:23 xxx systemd[1]: Started Logitech Configuration Daemon.
May 12 14:13:23 xxx logid[10234]: terminate called after throwing an instance of 'ipcgull::connection_failed'
May 12 14:13:23 xxx logid[10234]:   what():  Cannot autolaunch D-Bus without X11 $DISPLAY
May 12 14:13:23 xxx systemd[1]: logid.service: Main process exited, code=dumped, status=6/ABRT
May 12 14:13:23 xxx systemd[1]: logid.service: Failed with result 'core-dump'.
```

`echo $DISPLAY` outputs `:0`

antnesswcm, 2023-06-25:
I have the same problem. Can someone help me

tejus, 2023-08-08:
I no longer have this issue after building from current master at 94f6dba.

closed 2023-08-08 by tejus