PixlOne/logiops #377: Large memory consumption in logid

Barabazs, 2023-05-12:
I noticed that logid seams to be hogging a lot of memory since I updated to the latest version.

OS: Ubuntu 22.04.2 LTS
kernel: 5.19.0-41-generic 
logid version: v0.3.1

[image]

PixlOne, 2023-05-12:
Seems to be a memory leak, could you post your config and logs?

Barabazs, 2023-05-12:
Here is my config:
```
devices: ({
  name: "Wireless Mouse MX Master 3";
  smartshift:
  {
      on: true;
      threshold: 10;
      default_threshold: 30;
  };
  dpi: 3100;
  })
```
Where are the logs for logid stored?

I did found this in the "Logs" application:
```
 9 mei 17:48:24 kernel: Out of memory: Killed process 1996170 (logid) total-vm:23991604kB, anon-rss:20345592kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:41456kB oom_score_adj:0
 9 mei 11:48:51 kernel: Out of memory: Killed process 1339425 (logid) total-vm:14030132kB, anon-rss:13010196kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:26504kB oom_score_adj:0
 8 mei 15:27:28 kernel: Out of memory: Killed process 2308625 (logid) total-vm:22222132kB, anon-rss:17266928kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:35940kB oom_score_adj:0
```

PixlOne, 2023-05-12:
Run `sudo journalctl -u logid` to get the logs

Barabazs, 2023-05-12:
I grepped for today and I have 15204 lines which consists of 15051 lines saying `[INFO] /dev/hidraw3:1 woke up.`

Tail of the log:
```
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:33 nan logid[463433]: [INFO] /dev/hidraw3:1 woke up.
mei 12 10:12:47 nan logid[463433]: [INFO] /dev[ERROR] Failed to add device 1 to receiver on /dev/hidraw3: sendReport write failed: Broken pipe
mei 12 13:23:09 nan systemd[1]: Stopping Logitech Configuration Daemon...
mei 12 13:23:10 nan systemd[1]: logid.service: Deactivated successfully.
mei 12 13:23:10 nan systemd[1]: Stopped Logitech Configuration Daemon.
mei 12 13:23:10 nan systemd[1]: logid.service: Consumed 1min 23.265s CPU time.
mei 12 13:23:10 nan systemd[1]: Started Logitech Configuration Daemon.
```

I restarted the service at 13:23.

PixlOne, 2023-05-12:
Can you try out the fix-wakeup-memleak branch and let me know if that fixes it? I believe I've determined the issue.

PixlOne, 2023-05-16:
Can you try out the latest commit on that branch? @Barabazs I'm having trouble reproducing this bug personally.

PixlOne, 2023-05-19:
I'm going to merge this for now and close the issue, please re-open if this persists.

closed 2023-05-19 by PixlOne

Barabazs, 2023-05-19:
Hey @PixlOne I just wanted to let you know that it's still running smoothly.

PR #380 (2023-05-19): Fix memory leak on receiver waiting
Fixes #377