PixlOne/logiops #385: logid fails to start with device timeout. times out immediately. works randomly

rayzorben, 2023-05-26:
When logid runs it will hang at `timed out, waiting for input from device to initialize.`. It never proceeds. I re-run it 2-5 times and it starts working. This is an issue because running in a service I never know when it is going to work other than to just keep restarting it.

Here is where it doesn't work (note that the timed out prints IMMEDIATELY.
```
sudo logid -v
[WARN] Error at line 17: SettingTypeException
[WARN] Error at line 22: SettingTypeException
[DEBUG] Unsupported device /dev/hidraw7 ignored
[DEBUG] Ignoring virtual node on /dev/hidraw2
[INFO] Detected receiver at /dev/hidraw1
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] /dev/hidraw1:1 timed out, waiting for input from device to initialize.
```

Here is where it works:
```
sudo logid -v
[WARN] Error at line 17: SettingTypeException
[WARN] Error at line 22: SettingTypeException
[DEBUG] Unsupported device /dev/hidraw7 ignored
[DEBUG] Ignoring virtual node on /dev/hidraw2
[INFO] Detected receiver at /dev/hidraw1
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] /dev/hidraw1:1 timed out, waiting for input from device to initialize.
```

markozajc, 2023-08-29:
I have the same issue with a M650. Additionally, unplugging and replugging the bolt dongle or turning the mouse on and off while logid is running fixes it.

isalgueiro, 2023-10-25:
Same here with an MX Master 3. Can confirm the workaround from @markozajc works for me as well.