PixlOne/logiops #387: logid seems completely broken

hendriktews, 2023-05-28:
Hi,

with the following config
```
devices: ({
  name: "Wireless Mouse MX Master 3";

  buttons: (
    { cid: 0xc3;
      action = {
        type: "Keypress";
        keys: ["BTN_MIDDLE"];
      };
    }
  );
});
```
with the current version v0.3.2 `sudo ./logid -v -c ~/logid.cfg` outputs
```
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw0:255
[DEBUG] /dev/hidraw0:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```
and terminates. This config works fine in v0.2.4. With version v0.3.1 logid does not terminate, but the thumb button does not work as expected.

hendriktews, 2023-05-28:
The difference in behavior between v0.3.1 and v0.3.2 seems to come from be5ee9f7934a9a8b8443c9590bc45c41ebb4a844.
I also tried to bisect where the problem was introduced after v0.2.4, but I hit too many intermediate versions with other problems.

S-trace, 2023-05-29:
Yeah, such breaking changes are way too bad for bisect, I also hit this trouble while debugging other problem. The code should be buildable and runnable at each commit in master branch,

MKuranowski, 2023-05-29:
This issue still persists on v0.3.2 after installing the dbus config file (`make install`)

MKuranowski, 2023-05-29:
I have the same finding, 0.2.4 works fine; 0.3.1 and 0.3.2 no (0.3.0 does not build). git bisect suggest that the bug was introduced in 918ea63755a66d69373d7da431b61203fa878b41.

hendriktews, 2023-06-06:
Thanks for the feedback!
The debian packages 0.3.1 and 0.3.2-1 work for me, which suggests to me that I missed the installation. I will try make install later.

MKuranowski, 2024-03-15:
This is still an issue on v0.3.3