PixlOne/logiops #388: MX Master 3S problem at arch linux with bolt receiver

thanasistrisp, 2023-05-29:
I am running `sudo logid -v` and I get the following:
```
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[INFO] Detected receiver at /dev/hidraw2
[DEBUG] Unsupported device /dev/hidraw3 ignored
```
I have already paired with solaar.

MacielFede, 2023-06-02:
I have the same problem, with the same mice. I think they don't support the exact receiver we have. 
Anyway, if I connect it through Bluetooth the device is recognized and my configs are applied nicely, that's the solution by now.

TroyDowling, 2023-06-08:
Agreeing with @MacielFede. I went through the same thing when I set this mouse up; its support is still marginal. Bluetooth works great, Bolt not so much.

bambalamm, 2023-07-13:
The bolt received support in [version 0.3.0](https://github.com/PixlOne/logiops/releases/tag/v0.3.0).

For me (ubuntu 22.04, logid v0.3.3) the device was recognized _either_ after 20 minutes of scrolling through the issues _or_ after cycling the 'easy-switch' button 1-2-3 on the bottom of the mouse. Probably the latter.

```
$ sudo systemctl status logid
● logid.service - Logitech Configuration Daemon
Loaded: loaded (/lib/systemd/system/logid.service; enabled; vendor preset: enabled)
Active: active (running) since Thu 2023-07-13 17:22:43 CEST; 19min ago
Main PID: 51760 (logid)
Tasks: 8 (limit: 18927)
Memory: 1.4M
CPU: 26ms
CGroup: /system.slice/logid.service
└─51760 /usr/local/bin/logid

Jul 13 17:22:43 ubuntu systemd[1]: Started Logitech Configuration Daemon.
Jul 13 17:22:43 ubuntu logid[51760]: [INFO] Detected receiver at /dev/hidraw4
Jul 13 17:40:18 ubuntu logid[51760]: [INFO] Device found: MX Master 3S on /dev/hidraw4:2
```

shuhaowu, 2023-09-05:
I'm on v0.3.3 and also cannot detect my MX Master 3 for business.

```
$ sudo ./logid
[INFO] Config file does not exist, using empty config.
[INFO] Device found: Bluetooth Multi-Device Keyboard K380 on /dev/hidraw1:255
[WARN] Error adding device /dev/hidraw1: Device timed out
[INFO] Detected receiver at /dev/hidraw4
```

vedshastry, 2023-09-08:
I am on v0.3.3 too, and have a similar issue!

`~ % sudo logid -v                                                                                                               [0]
[INFO] Config file does not exist, using empty config.
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[INFO] Detected receiver at /dev/hidraw3
[DEBUG] Unsupported device /dev/hidraw0 ignored
.`

TroyDowling, 2023-09-08:
I too get that error, but I don't think that's relevant here. Consider pointing logid to the expected config file. This works from the terminal directly, or set up through a systemd service file.
```
$ sudo logid -c /etc/logid/logid.cfg
[INFO] Device found: Yeti X on /dev/hidraw6:255
[INFO] Detected receiver at /dev/hidraw2
[INFO] Device found: MX Master 3S on /dev/hidraw5:255
```

KGrewal1, 2023-09-29:
Has anyone found a solution to this: getting
```
[DEBUG] Unsupported device /dev/hidraw8 ignored
[DEBUG] Unsupported device /dev/hidraw9 ignored
[DEBUG] Unsupported device /dev/hidraw10 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
[DEBUG] Unsupported device /dev/hidraw11 ignored
[DEBUG] Unsupported device /dev/hidraw12 ignored
[INFO] Detected receiver at /dev/hidraw13
[DEBUG] Unsupported device /dev/hidraw4 ignored
```
detecting the bolt receiver but not the 3S connected through it

meekah7, 2023-11-20:
> Has anyone found a solution to this: getting
>
> ```
> [DEBUG] Unsupported device /dev/hidraw8 ignored
> [DEBUG] Unsupported device /dev/hidraw9 ignored
> [DEBUG] Unsupported device /dev/hidraw10 ignored [...]

press the button on the bottom of the mouse 3 times, until the 1 lights up again. seems like this is the only way to do it right now with the bolt reciever. apparently bluetooth works better but I can't try that either right now.

meekah7, 2023-11-20:
I think this thread is a duplicate of #385

kstoykov, 2024-02-21:
No matter how many times I press the button on the bottom of the mouse, it just cannot find it. Only the receiver is detected.

TomasLiutvinas, 2025-07-14:
try moving mouse while running the command or adding extended timeout, I encountered similar issue today.

```
io_timeout: 60000.0; # More time to detect the mouse
devices: (
  {
    name: "MX Master 3S";

<...restofconfig...>
```