PixlOne/logiops #389: Crashes with failed assertion when started on boot by systemd

MithicSpirit, 2023-06-04:
Enabling the logid service for systemd causes it to crash on system start with
```
logid: /build/logiops/src/logiops-v0.3.2/src/logid/backend/raw/IOMonitor.cpp:78: logid::backend::raw::IOMonitor::io_lock::~io_lock(): Assertion `ret != -1' failed.
```
Restarting the service manually afterwards or executing it directly fixes this issue.

SelfRef, 2023-06-11:
Tested on latest git version:

SelfRef, 2023-06-11:
Tested on latest git version with MX Ergo on Bluetooth:

```
cze 11 11:10:42 AE-L-AZE-G14 systemd[1]: Started Logitech Configuration Daemon.
cze 11 11:10:55 AE-L-AZE-G14 logid[1106]: logid: /home/tm/.cache/yay/logiops-git/src/logiops/src/logid/backend/raw/IOMonitor.cpp:78: logid::backend::raw::IOMonitor::io_lock::~io_lock(): Assertion `ret != -1' failed.
cze 11 11:10:55 AE-L-AZE-G14 systemd[1]: logid.service: Main process exited, code=dumped, status=6/ABRT
cze 11 11:10:55 AE-L-AZE-G14 systemd[1]: logid.service: Failed with result 'core-dump'.
cze 11 11:13:29 AE-L-AZE-G14 systemd[1]: Started Logitech Configuration Daemon.
cze 11 11:13:32 AE-L-AZE-G14 logid[6381]: [INFO] Device found: MX Ergo Multi-Device Trackball  on /dev/hidraw3:255
```

What's important - only the first run fails, the boot. After restart unit it works fine (last two lines). I assume it may something with access to resources.

SiyangShao, 2023-06-17:
Faced the same problem on arch linux. The problem seems not only be caused by the assertion, as when I just simply comment it and recompile the logid, the logid still nots work

mariuszste, 2023-07-07:
Can confirm this happening on boot. Configuring auto restart in systemd can be used as a temporary workaround.

> as when I just simply comment it and recompile the logid, the logid still nots work

That's exactly what assertions are for. They assert an impossible state and prevent unexpected behaviour down the line. Removing them is never the solution :rofl:

PixlOne, 2023-07-12:
This should be fixed in the latest version.

mariuszste, 2023-07-18:
Can confirm, haven't seen a single crash since the update.

closed 2023-07-20 by MithicSpirit