PixlOne/logiops #391: Does not detect Logitech MX Master 3S

ehsan-ali-th, 2023-06-06:
Does not detect Logitech MX Master 3S. I just get this:

root@debian-pc1:/home# logid
[INFO] Detected receiver at /dev/hidraw4
[INFO] Detected receiver at /dev/hidraw8
root@debian-pc1:/home#

davorono, 2023-06-06:
MX Master 3S [not listed](https://github.com/PixlOne/logiops/blob/main/TESTED.md) on compatible devices at the moment. 

But running `sudo logid` gives me a similar response and I'm running a MX Master 3 (not S), I'd just proceed with setup as normal.

ehsan-ali-th, 2023-06-07:
It seems it now detects my Logitech MX Master 3S. I added the following in my config:

devices: (
  {
    // For the 3S, the name is changed to "MX Master 3S"
    name: "MX Master 3S";
     // ..........
     // Button config is omitted 
   }
);

The issue now is that I don't know which button has what CID value.

root@debian-pc1:/home/# logid -v
[DEBUG] Ignoring virtual node on /dev/hidraw6
[DEBUG] Ignoring virtual node on /dev/hidraw7
[INFO] Detected receiver at /dev/hidraw5
[DEBUG] Unsupported device /dev/hidraw8 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[INFO] Detected receiver at /dev/hidraw4
[DEBUG] /dev/hidraw4:2 timed out, waiting for input from device to initialize.
^C
root@debian-pc1:/home# logid -v
[DEBUG] Ignoring virtual node on /dev/hidraw6
[DEBUG] Ignoring virtual node on /dev/hidraw7
[INFO] Detected receiver at /dev/hidraw5
[DEBUG] Unsupported device /dev/hidraw8 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[INFO] Detected receiver at /dev/hidraw4
[INFO] Device found: MX Master 3S on /dev/hidraw4:2
[DEBUG] /dev/hidraw4:2 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)

ehsan-ali-th, 2023-06-07:
OK I got this working now. I confirm "Logitech MX Master 3S" working 100% using this project.

closed 2023-06-07 by ehsan-ali-th

mrondr, 2024-01-23:
Can I ask you what did you do to have logid detect your 3s? I am stuck in a situation similar to your first post:

`sudo logid status`
only reports
`[INFO] Detected receiver at /dev/hidraw2`

ehsan-ali-th, 2024-01-24:
> Can I ask you what did you do to have logid detect your 3s? I am stuck in a situation similar to your first post:
> 
> `sudo logid status` only reports `[INFO] Detected receiver at /dev/hidraw2`

edit /etc/logid.cfg with the following content:

`devices: (
{
  // For the 3S, the name is changed to "MX Master 3S"
  name: "MX Master 3S";

  timeout = 10000;

  smartshift:
  {
    on: true;
    threshold: 20;
    torque: 10;
  };

  hiresscroll:
  {
    hires: false;
    invert: false;
    target: false;
  };
  
  dpi: 400;
  
  thumbwheel:
  {
    divert: true;
    invert: false;

    left: {        
      mode: "OnInterval"
      pixels: 2
      action: { 
        type: "Keypress"
        keys: ["KEY_VOLUMEDOWN"]
      }
    }
    right: {
      mode: "OnInterval"
      pixels: 2
      action: { 
        type: "Keypress"
        keys: ["KEY_VOLUMEUP"] 
      }
    }
    
    tap: {
      type: "Keypress"
      keys: ["KEY_MUTE"]
    }

  };
  
  buttons: (
  {
    # Thumb button
    cid: 0xc3;
    action =
    {
      type: "Gestures";
      gestures: (
        {
          direction: "Up";
          mode: "OnRelease";
          action =
          {
            type: "Keypress";
            keys: ["KEY_LEFTCTRL", "KEY_UP"];
          };
        },
        {
          direction: "Down";
          mode: "OnRelease";
          action =
          {
            type: "Keypress";
            keys: ["KEY_LEFTCTRL", "KEY_DOWN"];
          };
        },
        {
          direction: "Left";
          mode: "OnRelease";
          action =
          {
            type: "CycleDPI";
            dpis: [1000, 2000, 3000, 4000, 5000, 6000, 8000];
          };
        },
        {
          direction: "Right";
          mode: "OnRelease";
          action =
          {
            type = "ToggleSmartshift";
          }
        },
        {
          direction: "None";
          mode: "NoPress";
        }
      );
    };
  },
  {
    # Button behind the scroll
    cid: 0xc4;
    action =
    {
      type: "ToggleSmartshift";
    };
  },
  {
    # Back button: Previous workspace
    cid: 0x53;
    action =
    {
      type: "Keypress";
      keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_LEFT"];
    };
  },
  {
    # Next button: Next workspace
    cid: 0x56;
    action =
    {
      type: "Keypress";
      keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_RIGHT"];
    };
  });
});
`

and then:

`$ systemctl restart logid`

Basically, For the 3S, the name is changed to "MX Master 3S". You need to change it to "MX Master 3S".

mrondr, 2024-01-24:
Ah, my bad, I misread your previous post and thought you edited the file after having logid "recognize" your mouse.

For future reference, my status is now that while `sudo logid status` still does not list my mx 3s, I can edit the mouse configuration (thanks to @ehsan-ali-th file above with the correct name).