PixlOne/logiops #394: CycleDPI no longer working

volatilemolotov, 2023-06-08:
```                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "CycleDPI";
                            dpis: [50, 500, 1000, 1500, 2000, 3000, 4000];
                        };
                    },
```

Stopped working recently. No changes in DPI. Other gestures are working fine. 
Device is MX Master 3

troydowling-exegin, 2023-06-08:
Reporting a similar problem. Recently upgraded logiops-git (1:0.2.3.r18.gdbe2b28-1 -> 1:0.3.2.r0.g5767aac-1) on Arch Linux. I use an MX Master 3S, with the top button (below the scroll wheel) bound.

`
// Top button
cid: 0xc4;
action = {
  type: "Gestures";
  gestures: (
    {
      direction: "None";
      mode: "OnRelease";
      action = {
        type: "CycleDPI";
        dpis: [400, 1200, 2200, 3000];
      }
    }
  );
};
`

TroyDowling, 2023-06-08:
Reporting a similar problem. Recently upgraded logiops-git (1:0.2.3.r18.gdbe2b28-1 -> 1:0.3.2.r0.g5767aac-1) on Arch Linux. I use an MX Master 3S, with the top button (below the scroll wheel) bound.

```
// Top button
cid: 0xc4;
action = {
  type: "Gestures";
  gestures: (
    {
      direction: "None";
      mode: "OnRelease";
      action = {
        type: "CycleDPI";
        dpis: [400, 1200, 2200, 3000];
      }
    }
  );
};
```

my4ng, 2023-08-07:
```
// Top button
{
  cid: 0xc4;
  action = {
    type: "Gestures";
    gestures: (
      {
        direction: "None";
        mode: "OnRelease";
            action =
            {
              type: "CycleDPI";
              dpis: [1600, 2000, 2400, 3000];
            };
      }
    );
  };
}
``` 
Basically the same issue here, MX Master 3S, version 0.3.3.

bartoszp1992, 2023-12-11:
Same issue. Tested on MX master 2S and lift. 0xc4 working good e.g. as left control.

`            cid: 0xc4;
            action =
            {
		type: "CycleDPI";
        	dpis: [400, 600, 800, 1000, 1200, 1400, 1600];
            };`

notlaast, 2024-07-12:
I've had this same issue for many months now with my MX Master 3 on Pop OS 22.04, just confirmed it's still broken on v0.3.4.

I've prefer the non-gesture-based method of simple clicks to cycle:
`cid: 0x0c3;
action = {
	type: "CycleDPI";
	dpis: [400, 500, 800, 1500];
};`

but I've also tried the gesture-based method, as well as testing using the button below the wheel (i.e. `cid: 0xc4`). I can confirmed that both buttons support other functionality if reconfigured to do so; it really does seem like CycleDPI is simply broken somehow.

I've tried to fix it myself but my c++ is too poor and this code project is too complex for me to try to fix it :(