PixlOne/logiops #398: KEY_PLAYPAUSE not working

soype, 2023-06-28:
Hi, I've recently reinstalled everything on my linux PC and I can't get the "KEY_PLAYPAUSE" command to work on my gestures. All the other commands work. It does work on the keyboard.

Any ideas on what I should change?

Debug:
```

-event25  KEYBOARD_KEY            +1.742s	KEY_PLAYPAUSE (164) pressed
 event25  KEYBOARD_KEY            +1.742s	KEY_PLAYPAUSE (164) released
-event24  KEYBOARD_KEY            +1.742s	KEY_PLAYPAUSE (164) pressed
 event24  KEYBOARD_KEY            +1.742s	KEY_PLAYPAUSE (164) released
-event23  KEYBOARD_KEY            +1.742s	KEY_PLAYPAUSE (164) pressed
 event23  KEYBOARD_KEY            +1.742s	KEY_PLAYPAUSE (164) released
-event7   KEYBOARD_KEY            +2.486s	KEY_PLAYPAUSE (164) pressed
 event7   KEYBOARD_KEY            +2.586s	KEY_PLAYPAUSE (164) released

```

My config
```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: false;
        threshold: 30;
        torque: 50;
    };
    hiresscroll:
    {
        hires: true;
        invert: true;
        target: false;
    };
    dpi: 1200;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_UP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_DOWN"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_PLAYPAUSE"];
                        }
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "CycleDPI";
                            dpis: [400, 600, 800, 1000, 1200, 1400, 1600];
                        };
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type: "Keypress";
                keys: ["KEY_A"];
            };
        }
    );
}
);
```

S-trace, 2023-06-28:
Please try this config snippet:
```

soype, 2023-06-28:
Thanks. I reinstalled PopOS due to another reason and this config worked.

closed 2023-06-28 by soype