PixlOne/logiops #402: But in waitForDevice (when having two devices)

xinix00, 2023-07-26:
time out function not working properly.

1. _ready calls _addHandler
2. _addHandler calls addDevice
3. addDevice calls waitForDevice 
4. >> SECOND device joins (got keyboard and mouse) guess (does step 1 to 3) think releases the _wait_mutex for device 1
5. _addHandler continues and REMOVES the _waiters.erase(event.index) leaving it with NO waiters to wake up!

Currently fixed it for myself by skipping the _wait_mutex and do it the other way around:

waitForDevice will call _addHandler when finished, so it seems sufficient to do this:
        _waiters.erase(event.index);
        addDevice(event);

in stead of this:
        addDevice(event);
        const std::lock_guard lock(_wait_mutex);
        _waiters.erase(event.index);

I shall explain: 
_ready calls _addHandler, erases a non existing waiter.
 _addHandler calls addDevice, which will call waitForDevice, which is awaiting a call to _addHandler, so when finished it will ALWAYS erase the _waiters.erase(index)

P.s. so you can remove:
const std::lock_guard lock(_wait_mutex);
and all references to _wait_mutex completely

Maybe you want to change this, I saw a few people having the same problem.

azisaka, 2023-12-11:
The fix works! Just made the tests here: 

```
logiops/build$ sudo logid -v
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[INFO] Detected receiver at /dev/hidraw2
[DEBUG] /dev/hidraw2:2 timed out, waiting for input from device to initialize.
[DEBUG] /dev/hidraw2:3 timed out, waiting for input from device to initialize.
[INFO] Device found: MX Mechanical Mini on /dev/hidraw2:2
[DEBUG] /dev/hidraw2:2 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x34 |         |         |            |
[DEBUG] 0xd1 |         | YES     |            |
[DEBUG] 0xd2 |         | YES     |            |
[DEBUG] 0xd3 |         | YES     |            |
[DEBUG] 0xd4 | YES     | YES     |            |
[DEBUG] 0xde |         | YES     |            |
[DEBUG] 0xe2 | YES     | YES     |            |
[DEBUG] 0xe3 | YES     | YES     |            |
[DEBUG] 0xe7 | YES     | YES     |            |
[DEBUG] 0xe8 | YES     |         |            |
[DEBUG] 0xe9 | YES     |         |            |
[DEBUG] 0x103 | YES     | YES     |            |
[DEBUG] 0x108 | YES     | YES     |            |
[DEBUG] 0x10a | YES     | YES     |            |
[DEBUG] 0x10b | YES     |         |            |
[DEBUG] 0x10c | YES     |         |            |
[DEBUG] 0x10d | YES     |         |            |
[DEBUG] 0x10e | YES     |         |            |
[DEBUG] 0x10f | YES     |         |            |
[DEBUG] 0x110 | YES     |         |            |
[DEBUG] 0x111 | YES     |         |            |
[DEBUG] 0x112 | YES     |         |            |
[DEBUG] 0x115 | YES     |         |            |
[DEBUG] 0x117 | YES     |         |            |
[DEBUG] 0x118 | YES     |         |            |
[DEBUG] 0x119 | YES     |         |            |
[DEBUG] 0x11a | YES     |         |            |
[DEBUG] 0x11b | YES     |         |            |
[DEBUG] 0x11c | YES     | YES     |            |
[DEBUG] 0x11e | YES     |         |            |
[DEBUG] 0x13c | YES     |         |            |
[DEBUG] 0x141 | YES     | YES     |            |
[INFO] Device found: MX Master 3S on /dev/hidraw2:3
[DEBUG] /dev/hidraw2:3 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        |
[DEBUG] 0x51 |         |         | YES        |
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES
[DEBUG] Thumb wheel resolution: native (18), diverted (120)

```

spoutn1k, 2024-02-10:
This also works for me. Has this been made into a PR ? Is the project abandonned ?

ankon, 2024-06-07:
FWIW: This also seems to fix things for a "Wave Keys" keyboard:

```
[INFO] Config file does not exist, using empty config.
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[INFO] Detected receiver at /dev/hidraw3
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw9 ignored
[DEBUG] /dev/hidraw3:1 timed out, waiting for input from device to initialize.
[INFO] Device found: Wave Keys on /dev/hidraw3:1
[DEBUG] /dev/hidraw3:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0xd1 |         | YES     |            | 
[DEBUG] 0xd2 |         | YES     |            | 
[DEBUG] 0xd3 |         | YES     |            | 
[DEBUG] 0xde |         |         |            | 
[DEBUG] 0xe0 | YES     | YES     |            | 
[DEBUG] 0xe7 | YES     | YES     |            | 
[DEBUG] 0xe8 | YES     | YES     |            | 
[DEBUG] 0xe9 | YES     | YES     |            | 
[DEBUG] 0x103 | YES     | YES     |            | 
[DEBUG] 0x108 | YES     | YES     |            | 
[DEBUG] 0x10a | YES     | YES     |            | 
[DEBUG] 0x117 | YES     |         |            | 
[DEBUG] 0x118 | YES     |         |            | 
[DEBUG] 0x119 | YES     |         |            | 
[DEBUG] 0x11a | YES     |         |            | 
[DEBUG] 0x11b | YES     |         |            | 
[DEBUG] 0x11c | YES     | YES     |            | 
[DEBUG] 0x141 | YES     | YES     |            | 
```

reneas, 2025-01-27:
This worked for me as well only one small thing is bothering me: When running `logid -v` the command never stops, only if i stop the process manually through `ctrl +c` or `htop`