PixlOne/logiops #403: [Bug] Veritical Scrolling is too slow on Bluetooth

SubhadeepJasu, 2023-08-06:
Recently (likely after a kernel update) the vertical scrolling is too slow when I am using Bluetooth. The problem goes away when I use the Bolt receiver. It is so slow that one full swipe down of the scroll wheel doesn't move the page at all. I have to flick the scroll wheel to move the page. The horizontal scroll wheel still works as expected though.

ruionwriting, 2023-08-07:
I have the same issue (`6.2.0-26-generic`).

ThijsCas, 2023-08-09:
I had the same issue, but I fixed it by adding the following to /etc/logid.cfg:
`  hiresscroll:
  {
    hires: true;
    invert: false;
    target: false;
  };`

I did also run `make install` on the latest version, but I'm not sure if that was needed.

SubhadeepJasu, 2023-08-09:
Thanks will try this out. However it would be best to sort it out in the code.

u-ways, 2023-08-23:
I am having the same issue, I tried @ThijsCas suggestion and it did not work?

SubhadeepJasu, 2023-09-16:
[image]

Managed to fix it using LogiOpsGUI (Solaar).

I'll still keep the issue open, because it needs to be fixed from the driver defaults.

SubhadeepJasu, 2023-09-16:
Managed to fix it using LogiOpsGUI (Solaar)
[image]

I'll still keep the issue open because it needs to be fixed from the driver defaults