PixlOne/logiops #406: [WARN] Error adding device /dev/hidraw2: Device timed out - MX Keys for Business

IgorAntunes66, 2023-09-18:
Description:
When running the command "sudo logid", it gives an error identifying the MX Keys for Business keyboard model "YR0073".

Steps to Reproduce:
Run the command "sudo logid"

Evidence: Terminal Output

"[INFO] Device found: MX Master 3 for Business on /dev/hidraw1:255
[INFO] Device found: MX Keys for Business on /dev/hidraw2:255
[WARN] Error adding device /dev/hidraw2: Device timed out"

The mouse is detected normally, but the keyboard encounters this problem.

IgorAntunes66, 2023-09-18:
I run the commando "sudo logid -v" and get this informations

"[DEBUG] Unsupported device /dev/hidraw0 ignored
[INFO] Device found: MX Master 3 for Business on /dev/hidraw1:255
[DEBUG] /dev/hidraw1:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
[INFO] Device found: MX Keys for Business on /dev/hidraw2:255
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x0a | YES     |         |            | 
[DEBUG] 0x34 |         |         |            | 
[DEBUG] 0x6f | YES     |         |            | 
[DEBUG] 0xc7 | YES     | YES     |            | 
[DEBUG] 0xc8 | YES     | YES     |            | 
[DEBUG] 0xd1 | YES     |         |            | 
[DEBUG] 0xd2 | YES     |         |            | 
[DEBUG] 0xd3 | YES     |         |            | 
[DEBUG] 0xde |         | YES     |            | 
[DEBUG] 0xe2 | YES     | YES     |            | 
[DEBUG] 0xe3 | YES     | YES     |            | 
[DEBUG] 0xe4 | YES     | YES     |            | 
[DEBUG] 0xe5 | YES     | YES     |            | 
[DEBUG] 0xe6 | YES     | YES     |            | 
[DEBUG] 0xe7 | YES     | YES     |            | 
[DEBUG] 0xe8 | YES     | YES     |            | 
[DEBUG] 0xe9 | YES     |         |            | 
[DEBUG] 0xea | YES     |         |            | 
[DEBUG] 0xeb | YES     |         |            | 
[DEBUG] 0xec | YES     |         |            | 
[DEBUG] 0x103 | YES     | YES     |            | 
[DEBUG] 0x108 | YES     | YES     |            | 
[DEBUG] 0x10a | YES     |         |            | 
[DEBUG] 0x11c | YES     | YES     |            | 
[WARN] Error adding device /dev/hidraw2: Device timed out"

IgorAntunes66, 2023-09-22:
Anyone?

liamhays, 2023-09-24:
I'm having this issue too, with a M720 Triathlon connected over Bluetooth. `logid` detects the mouse on `/dev/hidraw1`, but it times out trying to add it. 

`fuser /dev/hidraw1` lists no processes using the file, and I don't have Piper or Solaar installed.

johannesrave, 2023-10-14:
i believe i have a similar issue:

```
[INFO] Device found: MX Mechanical Mini on /dev/hidraw2:255
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x34 |         |         |            | 
[DEBUG] 0xd1 |         | YES     |            | 
[DEBUG] 0xd2 |         | YES     |            | 
[DEBUG] 0xd3 |         | YES     |            | 
[DEBUG] 0xd4 | YES     | YES     |            | 
[DEBUG] 0xde |         | YES     |            | 
[DEBUG] 0xe2 | YES     | YES     |            | 
[DEBUG] 0xe3 | YES     | YES     |            | 
[DEBUG] 0xe7 | YES     | YES     |            | 
[DEBUG] 0xe8 | YES     |         |            | 
[DEBUG] 0xe9 | YES     |         |            | 
[DEBUG] 0x103 | YES     | YES     |            | 
[DEBUG] 0x108 | YES     | YES     |            | 
[DEBUG] 0x10a | YES     | YES     |            | 
[DEBUG] 0x10b | YES     |         |            | 
[DEBUG] 0x10c | YES     |         |            | 
[DEBUG] 0x10d | YES     |         |            | 
[DEBUG] 0x10e | YES     |         |            | 
[DEBUG] 0x10f | YES     |         |            | 
[DEBUG] 0x110 | YES     |         |            | 
[DEBUG] 0x111 | YES     |         |            | 
[DEBUG] 0x112 | YES     |         |            | 
[DEBUG] 0x115 | YES     |         |            | 
[DEBUG] 0x117 | YES     |         |            | 
[DEBUG] 0x118 | YES     |         |            | 
[DEBUG] 0x119 | YES     |         |            | 
[DEBUG] 0x11a | YES     |         |            | 
[DEBUG] 0x11b | YES     |         |            | 
[DEBUG] 0x11c | YES     | YES     |            | 
[DEBUG] 0x11e | YES     |         |            | 
[DEBUG] 0x13c | YES     |         |            | 
[DEBUG] 0x141 | YES     | YES     |            | 
[WARN] Error adding device /dev/hidraw2: Device timed out

```

wprzytula, 2023-11-07:
An analogous situation with MX Keys Mini.

wprzytula, 2023-11-07:
I've issued a pull request that resolves the problem.

deoxyseia, 2023-11-16:
@wprzytula I built your code but the problem persists, in this case with a MX Mechanical Mini. Here the output:

[INFO] Config file does not exist, using empty config.
[INFO] Device found: MX Master 3S on /dev/hidraw2:255
[INFO] Device found: MX Mechanical Mini on /dev/hidraw1:255
[WARN] Error adding device /dev/hidraw1: Device timed out

Some system details:
- Pop!_OS 22.04 LTS, based on Ubuntu Jammy

wprzytula, 2023-11-17:
> @wprzytula I built your code but the problem persists, in this case with a MX Mechanical Mini. Here the output:
>
> [INFO] Config file does not exist, using empty config. [INFO] Device found: MX Master [...]

Please post your output when run with `-v rawreport`. I'll investigate it.

liamhays, 2023-11-20:
My M720 Triathlon is also still timing out with your code. When it's connected over Bluetooth, logid crashes with the same timeout. `-v rawreport` output:

```
[DEBUG] Unsupported device /dev/hidraw0 ignored
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[DEBUG] Failed to add device /dev/hidraw1 on try 1, backing off for 500ms
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 02 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 02 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 03 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 03 02 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 03 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 03 12 4d 37 32 30 20 54 72 69 61 74 68 6c 6f 6e 20 4d 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 03 12 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 03 12 75 6c 74 69 2d 44 65 76 69 63 65 20 4d 6f 75 73 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 03 12 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 03 12 65 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 12 00 00 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 12 04 05 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 02 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 02 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 03 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 03 02 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 03 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 03 12 4d 37 32 30 20 54 72 69 61 74 68 6c 6f 6e 20 4d 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 03 12 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 03 12 75 6c 74 69 2d 44 65 76 69 63 65 20 4d 6f 75 73 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 03 12 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 03 12 65 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[INFO] Device found: M720 Triathlon Multi-Device Mouse on /dev/hidraw1:255
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 02 22 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 IN:  11 ff 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 00 02 21 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[WARN] Error adding device /dev/hidraw1: Device timed out
```

When the mouse is connected via the Unifying receiver, logid (your branch, at least) works fine and I can remap the gesture thumb button.

wprzytula, 2023-11-20:
> My M720 Triathlon is also still timing out with your code. When it's connected over Bluetooth, logid crashes with the same timeout.
> When the mouse is connected via the Unifying receiver, logid (your branch, at least) works fine and I can remap the gesture thumb button.

Could you please provide your `-v rawreport` output for the mouse connected via the Unifying receiver? Maybe comparison of those could give some insight.

liamhays, 2023-11-20:
> Could you please provide your `-v rawreport` output for the mouse connected via the Unifying receiver? Maybe comparison of those could give some insight.
 
(M720 Triathlon over Bluetooth on Intel 7265D)

```
[DEBUG] Unsupported device /dev/hidraw0 ignored
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 02 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 02 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 03 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[DEBUG] Failed to add device /dev/hidraw3 on try 1, backing off for 500ms
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 02 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 02 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 03 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 03 02 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 03 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 03 12 4d 37 32 30 20 54 72 69 61 74 68 6c 6f 6e 20 4d 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 03 12 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 03 12 75 6c 74 69 2d 44 65 76 69 63 65 20 4d 6f 75 73 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 03 12 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 03 12 65 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 68 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 65 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 6c 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 12 00 00 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 12 04 05 6f 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 02 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 02 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 03 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 03 02 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 03 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 03 12 4d 37 32 30 20 54 72 69 61 74 68 6c 6f 6e 20 4d 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 03 12 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]
[INFO] Device found: M720 Triathlon Multi-Device Mouse on /dev/hidraw3:255
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 02 22 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 IN:  11 ff 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[RAWREPORT] /dev/hidraw3 OUT: 11 ff 00 02 21 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[WARN] Error adding device /dev/hidraw3: Device timed out
```

Between a few runs, sometimes there's more data, sometimes there's less, but it always times out.

wprzytula, 2023-11-20:
> > Could you please provide your `-v rawreport` output for the mouse connected via the Unifying receiver? Maybe comparison of those could give some insight.
> 
> (M720 Triathlon over Bluetooth on Intel 7265D)

You mentioned Bluetooth connection and I can see a timeout; and I asked about the Unifying receiver instead and a successful connection log, for comparison with the erroneous connection using Bluetooth.

liamhays, 2023-11-20:
I am a moron, it has been fixed. Sorry about that :smiley:

I also am definitely running your code, `ReprogControls.cpp` has `cid` as `uint16_t`.

deoxyseia, 2023-11-22:
@wprzytula, my bad, I built using your fork but using main instead the improved branch. This works well. 

Thanks for the support!

vbvr, 2023-12-19:
I have a POP Keys and POP Mouse, and encountered the same timeout error (logiops 0.3.3 on gentoo). Applying the patch from @wprzytula to my build resolved this, as well. Thank you.

PR #414 (2024-05-28): fix bug: represent cid as `uint16_t`, not `uint8_t`
Fixes: #406
The problem description is in the referenced issue.

## Diagnosis

The function `ReprogControlsV4::setControlReporting()` erroneously took `cid` as `uint8_t`. Because of that, reports contained only lower byte of any cid, so any ReprogControls request for such cid resulted in error response. Lack of error response handling in this library led to timeout.

E.g.:
```
[DEBUG] Configuring button: 0x103
[RAWREPORT] /dev/hidraw1 OUT: 11 ff 08 32 00 03 22 01 03 00 00 00 00 00 00 00 00 00 00 00
[RAWREPORT] /dev/hidraw1 IN:  11 ff ff 08 32 [... cut]
closed 2024-05-28 by PixlOne