PixlOne/logiops #407: compilation errors on Almalinux 8 / RH8

simonpr, 2023-10-11:
[  1%] Building CXX object src/ipcgull/CMakeFiles/ipcgull.dir/src/function.cpp.o
[  2%] Building CXX object src/ipcgull/CMakeFiles/ipcgull.dir/src/property.cpp.o
[  4%] Building CXX object src/ipcgull/CMakeFiles/ipcgull.dir/src/signal.cpp.o
[  5%] Building CXX object src/ipcgull/CMakeFiles/ipcgull.dir/src/interface.cpp.o
[  7%] Building CXX object src/ipcgull/CMakeFiles/ipcgull.dir/src/node.cpp.o
[  8%] Building CXX object src/ipcgull/CMakeFiles/ipcgull.dir/src/exception.cpp.o
[ 10%] Building CXX object src/ipcgull/CMakeFiles/ipcgull.dir/src/common_gdbus.cpp.o
[ 11%] Building CXX object src/ipcgull/CMakeFiles/ipcgull.dir/src/server_gdbus.cpp.o
[ 11%] Built target ipcgull
[ 13%] Linking CXX static library libipcgull.a
[ 13%] Built target ipcgull_static
[ 14%] Building CXX object src/logid/CMakeFiles/logid.dir/logid.cpp.o
[ 16%] Building CXX object src/logid/CMakeFiles/logid.dir/util/log.cpp.o
[ 17%] Building CXX object src/logid/CMakeFiles/logid.dir/config/config.cpp.o
[ 19%] Building CXX object src/logid/CMakeFiles/logid.dir/InputDevice.cpp.o
[ 20%] Building CXX object src/logid/CMakeFiles/logid.dir/DeviceManager.cpp.o
/home/simon/logiops/src/logid/DeviceManager.cpp: In member function ‘virtual void logid::DeviceManager::addDevice(std::__cxx11::string)’:
/home/simon/logiops/src/logid/DeviceManager.cpp:71:38: error: ‘class std::set<short unsigned int>’ has no member named ‘contains’
             config()->ignore.value().contains(raw_dev->productId())) {
                                      ^~~~~~~~
make[2]: *** [src/logid/CMakeFiles/logid.dir/build.make:132: src/logid/CMakeFiles/logid.dir/DeviceManager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:170: src/logid/CMakeFiles/logid.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

simonpr, 2023-10-11:
Solved this replacing all instances of contains with count

noworky, 2024-04-19:
I had the same problem, found the same fix.

Now facing this:
 50%] Linking CXX executable ../../logid
CMakeFiles/logid.dir/Configuration.cpp.o: In function `logid::Configuration::Configuration(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
Configuration.cpp:(.text+0xa9c): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
Configuration.cpp:(.text+0xaa4): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/logid/CMakeFiles/logid.dir/build.make:1010: logid] Error 1
make[1]: *** [CMakeFiles/Makefile2:170: src/logid/CMakeFiles/logid.dir/all] Error 2
make: *** [Makefile:136: all] Error 2