PixlOne/logiops #408: linking error - Almalinux8 / RH8

simonpr, 2023-10-11:
[ 50%] Linking CXX executable ../../logid
CMakeFiles/logid.dir/Configuration.cpp.o: In function `logid::Configuration::Configuration(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
Configuration.cpp:(.text+0xa9c): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
Configuration.cpp:(.text+0xaa4): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/logid/CMakeFiles/logid.dir/build.make:1010: logid] Error 1
make[1]: *** [CMakeFiles/Makefile2:170: src/logid/CMakeFiles/logid.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

simonpr, 2023-10-11:
Solved this by compiling outside of make adding -lstdc++fs to the compile string

yueyangyang, 2024-03-23:
> Solved this by compiling outside of make adding -lstdc++fs to the compile string

I am not familiar with C++. Can you provide specific instructions? Thank you