PixlOne/logiops #409: logiops silently exits with 0 with "Wireless Mouse MX Master 3"

ckiee, 2023-10-16:
The title, and the output from logid v0.3.3:
```
ckie@cookiemonster ~/git/nixpkgs -> sudo ./result/bin/logid -v; echo $?
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Ignoring virtual node on /dev/hidraw1
[INFO] Detected receiver at /dev/hidraw0
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
0
```

I ran a git bisect and this regression was introduced in https://github.com/PixlOne/logiops/commit/dbe24f9350212c1ea68dc2de60f431117214fb0e, huge diff!

There may also be other bugs afoot but I cannot tell if they were fixed because logiops does not launch on any of these v0.3.x releases.

johantiden, 2023-11-13:
Same here with 94f6dba
```
~/git/johantiden/logiops/build
❯ cmake -DCMAKE_BUILD_TYPE=Release ..
...
❯ sudo make install/local
~/git/johantiden/logiops/build
❯ sudo ./logid -v; echo $?
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Ignoring virtual node on /dev/hidraw4
[DEBUG] Ignoring virtual node on /dev/hidraw5
[DEBUG] Ignoring virtual node on /dev/hidraw6
[INFO] Detected receiver at /dev/hidraw3
[DEBUG] Unsupported device /dev/hidraw2 ignored
[DEBUG] Failed to add device /dev/hidraw8 on try 1, backing off for 500ms
0
```

AlexLJordan, 2023-12-13:
I have the same issue as OP after updating from 0.2.3
Gonna downgrade again, hope it's fixed eventually.

cjshearer, 2025-03-16:
What's worse is that the latest version "works" when running as root, so I spent the better part of a weekend chasing "permissions issues" and "the right `udev` incantations". I was comparing strace outputs between root and root-less runs, chasing down these red herring permissions issues, then it dawned on me that the program was exiting 0, so this was clearly a bug. 

Oh well...I guess running 0.2.3 is a 'fix', and I did learn some stuff, so ¯\_(ツ)_/¯

For what it's worth, here's what Copilot thinks the issue could be:
> Me: Why might this program exit 0 early when running as user when it doesn't as root? I expect it to keep running in both cases.
>
> Copilot: The program might exit with EXIT_SUCCESS early when running as a user but not as root due to the behavior of the setbuf function call in the main function. Specifically, the line:
>
 ```c++
setbuf(stdout, NULL);
```
> sets the buffer for stdout to NULL, which means that all output to stdout is immediately flushed. This behavior could potentially lead to different timing or interactions when the program is running as a non-root user compared to a root user, especially if the logging system or journald follows its logs differently based on user permissions.