PixlOne/logiops #416: MX Anywhere 3 scroll wheel very slow sensitive

JasonEleventeen, 2023-11-10:
Vertical wheel is not sensitive enough and scrolls very slow

```     hiresscroll:
    {
        hires: false;
        invert: false;
        target: true;

        up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 500.0;
        },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -500.0;
        },
    }; ```

JasonEleventeen, 2023-11-12:
Dug out the dongle, appears to only be an issue in bluetooth mode

dporta97, 2023-11-17:
In addiction to logid I have installed Solaar and this problem has been solved by applying the rotation speed in Solaar. It has kept all the shortcuts that I had been configured with logid

Staindk, 2023-11-20:
I think you need to use `REL_WHEEL` and not `REL_WHEEL_HI_RES`, but I'm unsure.

Guess I upgraded this package last week because now my Anywhere 3 also scrolls super slowly. It used to work just fine :cry: 

The following helps but mousewheel still seems quite laggy:

```
hiresscroll: {
    hires: false;
    invert: false;
    target: true;
    up: {
        mode: "Axis";
        axis: "REL_WHEEL";
        axis_multiplier: 1.0;
    },
    down: {
        mode: "Axis";
        axis: "REL_WHEEL";
        axis_multiplier: -1.0;
    }
  };
```