PixlOne/logiops #427: Scroll speed adjustment: REL_WHEEL_HI_RES scrolls up on 'down' gesture.

al1357, 2023-12-26:
Hi. I have a problem setting scroll speed on Debian 12, MX Master 2S, LogiOps v0.3.3 with `REL_WHEEL_HI_RES` axis, which previously worked on my Ubuntu install, I think it was LogiOps v0.2.4 there. Scroll down scrolls up instead of down. Here is my working Ubuntu config adjusted to v0.3.3(decimal axis_multiplier):

```
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: true;
	up: {
		mode: "Axis";
		axis: "REL_WHEEL_HI_RES";
		axis_multiplier: 2.0;
	}
	down: {
		mode: "Axis";
		axis: "REL_WHEEL_HI_RES";
		axis_multiplier: -2.0;
	}
    };
```
I either need to set `target: false` and resign from adjusting scroll speed or use `axis: "REL_WHEEL";`. This offers worse precision in setting axis_multiplier where 1.8 is several lines and 1.9 is more than half page - it's not terrible though. Example of functioning config:

```
hiresscroll:
{
        hires: true;
        invert: false;
        target: true;
        up: {
                mode: "Axis";
                axis: "REL_WHEEL";
                axis_multiplier: 1.8;
        }
        down: {
                mode: "Axis";
                axis: "REL_WHEEL";
                axis_multiplier: -1.8;
        }
}
```
Can you please suggest how to use `REL_WHEEL_HI_RES` in a way that wheel direction works correctly?

Sdghasemi, 2024-02-07:
I had the same issue on my MX Master 3S, since v0.2.4 had various bugs I used this PR to fix the problem on v3.3.0. Now everything works fine thanks to the author.
https://github.com/PixlOne/logiops/pull/413

paolomainardi, 2024-05-09:
@Sdghasemi how did you used it ? did you rebuild it your own version starting from there ?

markmb88, 2024-10-22:
> @Sdghasemi how did you used it ? did you rebuild it your own version starting from there ?

you can fork this repo, merge the PR and build/install. seems to have worked for my MX Master 3's sometimes unresponsive scroll wheel