PixlOne/logiops #428: `Device timed out` error when logi bolt is connected via thunderbolt 3 hub

alllexx88, 2024-01-07:
Hello,

I use a TB3 device (Razer Core X Chroma) with a usb 3.0 hub in it, and have a logi bolt connected to it. When I run `logid` I get this error:
```
$ sudo logid
[INFO] Detected receiver at /dev/hidraw6
[WARN] Error adding device /dev/hidraw6: Device timed out
```
If I connect the logi bolt to a laptop's usb port directly, the error goes away:
```
$ sudo logid
[INFO] Detected receiver at /dev/hidraw6
[INFO] Device found: MX Master 3S on /dev/hidraw6:2
```

It would be great to fix the issue. I'm using `logiops 0.3.3-1` on Arch installed via AUR: https://aur.archlinux.org/packages/logiops

Thank you!

kstoykov, 2024-02-21:
@alllexx88 How did you managed to get it running? I connect the logi bolt to my laptop and ./logid can only find the receiver. It cannot find any device.

forresthopkinsa, 2024-03-02:
Agree, I can't use a Bolt receiver through a USB hub, it only works when connected directly to the machine (which is a big problem on the growing number of laptops that have no USB-A ports!)

alllexx88, 2024-03-03:
@kstoykov I didn't really resolve it with logiops. Since I only need to change my mouse DPI, I'm using solaar instead, which works perfectly fine.