PixlOne/logiops #429: Multiple devices on the same receiver can not be detected

kheber, 2024-01-07:
When using a receiver with two devices connected to that same receiver, logid is unable to detect either device:
```
$ sudo logid -v
[DEBUG] Ignoring virtual node on /dev/hidraw2
[DEBUG] Ignoring virtual node on /dev/hidraw1
[INFO] Detected receiver at /dev/hidraw0
[DEBUG] Unsupported device /dev/hidraw3 ignored
```
Turning either device off does not remedy the issue. 

Interestingly, switching only one device to a bluetooth connection will make both devices recognized:

```
$ sudo logid -v
[DEBUG] Ignoring virtual node on /dev/hidraw2
[DEBUG] Ignoring virtual node on /dev/hidraw1
[INFO] Detected receiver at /dev/hidraw0
[DEBUG] Unsupported device /dev/hidraw3 ignored
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw0:2
[DEBUG] /dev/hidraw0:2 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
[INFO] Device found: MX Keys Wireless Keyboard on /dev/hidraw4:255
[DEBUG] /dev/hidraw4:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
```
In this case the keyboard was switched to bluetooth, and the mouse was on the receiver. When doing it the other way around (mouse on bluetooth, keyboard on receiver), the keyboard is recognized on `/dev/hidraw0:1` and the mouse is recognized on `/dev/hidraw5:255`.

alllexx88, 2024-01-08:
I can confirm I experience the same behaviour with an `MX Keys S` keyboard and an `MX Master 3S` mouse, logiops version `0.3.3`

glenl, 2024-04-07:
Just to add some characterization: 
Run `sudo logid -v` and while the command is running, continually tap the shift key, both devices will be recognized (`MX Keys S` and `MX Master 3S`).

````
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[INFO] Detected receiver at /dev/hidraw2
[INFO] Device found: MX Keys S on /dev/hidraw2:1
[DEBUG] /dev/hidraw2:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x0a | YES     |         |            |
[DEBUG] 0x6f | YES     |         |            |
[DEBUG] 0xc7 | YES     | YES     |            |
[DEBUG] 0xc8 | YES     | YES     |            |
[DEBUG] 0xd1 |         |         |            |
[DEBUG] 0xd2 |         |         |            |
[DEBUG] 0xd3 |         |         |            |
[DEBUG] 0xde |         |         |            |
[DEBUG] 0xe2 | YES     | YES     |            |
[DEBUG] 0xe3 | YES     | YES     |            |
[DEBUG] 0xe4 | YES     | YES     |            |
[DEBUG] 0xe5 | YES     | YES     |            |
[DEBUG] 0xe6 | YES     | YES     |            |
[DEBUG] 0xe7 | YES     | YES     |            |
[DEBUG] 0xe8 | YES     | YES     |            |
[DEBUG] 0xe9 | YES     |         |            |
[DEBUG] 0xea | YES     |         |            |
[DEBUG] 0x103 | YES     | YES     |            |
[DEBUG] 0x108 | YES     | YES     |            |
[DEBUG] 0x10a | YES     |         |            |
[DEBUG] 0x11c | YES     | YES     |            |
[DEBUG] /dev/hidraw2:1 timed out, waiting for input from device to initialize.
[INFO] Device found: MX Master 3S on /dev/hidraw2:2
[DEBUG] /dev/hidraw2:2 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        |
[DEBUG] 0x51 |         |         | YES        |
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
````
Devices can work in mysterious ways so I'm not going to embarrass myself by guessing why.

stevethemonk, 2024-04-27:
same problem here on ubuntu 20.04, tapping shift does nothing
sudo logid -v returns this
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[INFO] Detected receiver at /dev/hidraw3
[DEBUG] Unsupported device /dev/hidraw4 ignored

BUT 
sudo systemctl status logid , returns this
● logid.service - Logitech Configuration Daemon
     Loaded: loaded (/lib/systemd/system/logid.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2024-04-27 15:01:51 BST; 1min 1s ago
   Main PID: 2030 (logid)
      Tasks: 8 (limit: 38230)
     Memory: 3.3M
     CGroup: /system.slice/logid.service
             └─2030 /usr/bin/logid

Apr 27 15:01:51 steve systemd[1]: Started Logitech Configuration Daemon.
Apr 27 15:01:51 steve logid[2030]: [INFO] Detected receiver at /dev/hidraw3
Apr 27 15:01:51 steve logid[2030]: [INFO] Device found: MX Keys S on /dev/hidraw3:1
Apr 27 15:02:03 steve logid[2030]: [INFO] Device found: MX Master 3S on /dev/hidraw3:2

The only feature i care about is the thumb button switching between active windows, does anyone have any ideas