PixlOne/logiops #431: Receiver wasn't detected, I needed to pair/activate it on my windows boot, then reboot to linux it worked

thiolliere, 2024-01-28:
TLDR: Maybe the doc could notify that pairing the usb receiver on a windows computer is needed for the receiver to be detected?

I have Mouse MX master 3 for business.

Initially it only worked on bluetooth. When I inserted the small usb device it didn't work, no receiver were detected.

Then I went to windows (same computer different boot), plug the usb stuff, it started a pairing process with the logitech software, I had to click what was indicated, like left click, right click etc... then the pairing on windows was successful.

Then I went to linux, and everything worked:
command logid:
```
[INFO] Config file does not exist, using empty config.
[INFO] Detected receiver at /dev/hidraw2
```

Maybe the doc could notify that pairing the usb receiver on a windows computer is needed for the receiver to be detected?

Thanks a lot for logiops!

mtbrobotanist, 2024-02-07:
@thiolliere You can pair devices to a receiver on linux with this: https://github.com/Lekensteyn/ltunify

Flimm, 2024-05-20:
You can also pair devices on Linux using [Solaar](https://github.com/pwr-Solaar/Solaar), which offers a GUI. Both `ltunify` and `solaar` may be available in your Linux distro's repositories.

thiolliere, 2024-06-03:
I don't mind closing the issue, I mainly opened it for helping other who would run into the same issue as me.