PixlOne/logiops #435: Cannot run logid -v

ualonso011, 2024-02-08:
Hi,
When I try to run the command "sudo logid -v" it returns the following error:

terminate called after throwing an instance of 'std::runtime_error'
  what(): IOMonitor double run
Aborted

wolandmaster, 2024-05-14:
I have the same problem. Is there any solution for it?

felipeek, 2024-06-04:
just ran into the same issue.

In my case, the problem was my configuration file. I tried reducing the configuration file to simply:

```
devices: (
{
    name: "Tunable RGB Gaming Mouse G502";
    dpi: 1000;
}
);
```

But then logid failed to initialize with `what(): IOMonitor double run`

I now simply copied the [example configuration](https://raw.githubusercontent.com/PixlOne/logiops/main/logid.example.cfg) and replaced `Wireless Mouse MX Master` with `Tunable RGB Gaming Mouse G502` (which is my device) and logid was able to start. It is also working as if I change the dpi in the configuration it works.