PixlOne/logiops #444: Not working Master 3S - Ubuntu 22.04 (gnome)

fnagua, 2024-03-08:
Hi there! I've been using logiops in Linux mint for a while and everything was fantastic. Now I changed to Ubuntu 22.04 with gnome, and after doing the same as I did in the other OS now seems to be broken. I'm using the same file (and I know it's detecting the config because I'm able to change the DPI. But when assigning a button, it does not emit any events through ` libinput debug-events --show-keycodes`. Seems like the button I assign in the config, the button that stop sending events... Any clue?

Thank you in advance!

I leave here my config, just FYI:
```
devices: ({
  name: "MX Master 3S";

  dpi: 750; // max 4000

  // A lower threshold number makes the wheel switch to free-spin mode
  // quicker when scrolling fast.
  smartshift: { on: true; threshold: 255; };
  hiresscroll: {
    hires: false;
    invert: false;
    target: false;
  };

  // Buttons:
  // 0x50 -> Left
  // 0x51 -> Right
  // 0x52 -> Middle
  // 0x53 -> Back
  // 0x56 -> Forward
  // 0xc3 -> Thumb button (not wheel)
  // 0xc4 -> Top button
  // 0xd7 -> ¿¿Switch receivers??
  buttons: (
    // BACK BUTTON
    //{ cid: 0x53; action: { type: "Keypress"; keys: ["KEY_FORWARD"]; }; },

    // FORWARD BUTTON 
    // { cid: 0x56; action: { type: "Keypress"; keys: ["KEY_BACK"]; }; },
    // {
    //   cid: 0x56;
    //   action: {
    //     type: "Gestures";
    //     gestures: (
    //       {
    //         direction: "None";
    //         mode: "OnRelease";
    //         action: { type: "Keypress"; keys: ["KEY_BACK"]; };
    //       }
    //     );
    //   };
    // },

    // TOP BUTTON
    {
      cid: 0xc4;
      action: {
        type: "Gestures";
        gestures: (
          {  // CLICK -> PAUSE/PLAY
            direction: "None"; 
            mode: "OnRelease";
            action: { type: "Keypress"; keys: ["KEY_PLAYPAUSE"]; };
          },
          {  // RIGHT -> NEXT SONG
            direction: "Right"; 
            mode: "OnThreshold";
            threshold: 500;
            action: {type: "Keypress"; keys: ["KEY_NEXTSONG"]; };
          },
          {  // LEFT -> PREV SONG
            direction: "Left"; 
            mode: "OnThreshold";
            threshold: 500;
            action: {type: "Keypress"; keys: ["KEY_PREVIOUSSONG"]; };
          },
          {  // UP -> VOLUME UP
            direction: "Up";
            mode: "OnInterval";
            interval: 250;
            action: {type: "Keypress"; keys: ["KEY_VOLUMEUP"]; };
          },
          {  // DOWN -> VOLUME DOWN
            direction: "Down";
            mode: "OnInterval";
            interval: 250;
            action: {type: "Keypress"; keys: ["KEY_VOLUMEDOWN"]; };
          }
        );
      };
    },

    // THUMB BUTTON
    {
      cid: 0xc3;
      action: {
        type: "Gestures";
        gestures: (
          // {  // CLICK -> VIEW WORKSPACES (change to view windows??)
          //   direction: "None";
          //   mode: "OnRelease";
          //   action: { type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_DOWN"]; };
          // },
          {  // RIGHT -> MOVE TO RIGHT WORKSPACE
            direction: "Right";
            mode: "OnThreshold";
            threshold: 500;
            action: {type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_RIGHT"]; };
          },
          {  // LEFT -> MOVE TO LEFT WORKSPACE
            direction: "Left";
            mode: "OnThreshold";
            threshold: 500;
            action: {type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_LEFT"]; };
          },
          {  // UP -> VIEW WORKSPACES
            direction: "Up";
            mode: "OnThreshold";
            threshold: 500;
            action: {type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_UP"]; };
          },
          {  // DOWN -> VIEW APPS
            direction: "down";
            mode: "OnThreshold";
            threshold: 500;
            action: {type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_DOWN"]; };
          }
        );
      };
    }
  );

  // Divert is to handle thumb wheel with logiops. Then, you would be able to customize:
  // right, left, tap, touch, proxy (proximity)
  // I could not define horizontal scroll. I don't know the KEY_CODE
  thumbwheel: {
    divert: false;
    invert: true;

    // This needs to enable divert !!!
    right: {
      mode: "OnInterval"
      interval: 2;
        action: {
          type: "Keypress";
          keys: ["KEY_VOLUMEDOWN"];
        }
    };
    left: {
      mode: "OnInterval"
      interval: 2;
        action: {
          type: "Keypress";
          keys: ["KEY_VOLUMEUP"];
        }
    };

    // tap: {  // touch, proxy (proximity)
    //   type: "Keypress";
    //   keys: ["KEY_MUTE"];  // KEY_MUTE
    // };

  };
});
```

prashantgupta7775, 2024-03-21:
Hi, I am trying to setup the config , but the I am getting the error.
i followed this article to setup.
can you share the  location where you have placed config?

[INFO] Config file does not exist, using empty config.
[INFO] Detected receiver at /dev/hidraw5

fnagua, 2024-03-21:
@prashantgupta7775 Yes, sure thing.
I have the config file in `/etc/logid.cfg`

prashantgupta7775, 2024-03-21:
~root/logiops/etc/logid.cfg 

or

~root/logiops/build/etc/logid.cfg 

which one is yours I tried both arent working.

fnagua, 2024-03-21:
No no, the absolute path is that one, `/etc/logid.cfg`

prashantgupta7775, 2024-03-21:
I tried creating config at  /home/etc/logid.cfg. isnt working.
can you share some reference image or so?

fnagua, 2024-03-21:
[image]

But the config route should be the `/etc/logid.cfg`, not in home or root or whatever.

prashantgupta7775, 2024-03-21:
cool got it.
But it says, we cant paste here .
[image]

fnagua, 2024-03-21:
You need to do it with sudo

Try this:
- Create the config file in a regular directory, such as `~/logiops/logid.cfg`
- Copy this config file in the config route, with the command `sudo cp ~/logiops/logid.cfg /etc/logid.cfg`

prashantgupta7775, 2024-03-21:
Got it , config is detecting, Thanks for helping out.
but there is no change using it.
so there is any thing else need to do except sudo logid.

fnagua, 2024-03-22:
This is my problem, I don't know what am I missing, because I'm not able to make it work either... I was using it properly in Linux Mint, but now in ubuntu nope :(

fnagua, 2024-04-03:
I don't know what happened, but everything seems to be working now...

closed 2024-04-03 by fnagua