PixlOne/logiops #445: Compilation fails with LTO

ConiKost, 2024-03-16:
If LTO is enabled, compilation will fail. You can use ```-Werror=odr``` to test this.

```[67/67] : && /usr/bin/x86_64-pc-linux-gnu-g++ -O2 -march=x86-64 -pipe -pipe -frecord-gcc-switches -fno-diagnostics-color -fmessage-length=0 -flto -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wall -Wextra -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 src/ipcgull/CMakeFiles/ipcgull.dir/src/function.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/property.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/signal.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/interface.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/node.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/exception.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/common_gdbus.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/server_gdbus.cpp.o src/logid/CMakeFiles/logid.dir/logid.cpp.o src/logid/CMakeFiles/logid.dir/util/log.cpp.o src/logid/CMakeFiles/logid.dir/config/config.cpp.o src/logid/CMakeFiles/logid.dir/InputDevice.cpp.o src/logid/CMakeFiles/logid.dir/DeviceManager.cpp.o src/logid/CMakeFiles/logid.dir/Device.cpp.o src/logid/CMakeFiles/logid.dir/Receiver.cpp.o src/logid/CMakeFiles/logid.dir/Configuration.cpp.o src/logid/CMakeFiles/logid.dir/features/DPI.cpp.o src/logid/CMakeFiles/logid.dir/features/SmartShift.cpp.o src/logid/CMakeFiles/logid.dir/features/HiresScroll.cpp.o src/logid/CMakeFiles/logid.dir/features/RemapButton.cpp.o src/logid/CMakeFiles/logid.dir/features/DeviceStatus.cpp.o src/logid/CMakeFiles/logid.dir/features/ThumbWheel.cpp.o src/logid/CMakeFiles/logid.dir/actions/Action.cpp.o src/logid/CMakeFiles/logid.dir/actions/NullAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/KeypressAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/ToggleHiresScroll.cpp.o src/logid/CMakeFiles/logid.dir/actions/ToggleSmartShift.cpp.o src/logid/CMakeFiles/logid.dir/actions/CycleDPI.cpp.o src/logid/CMakeFiles/logid.dir/actions/ChangeDPI.cpp.o src/logid/CMakeFiles/logid.dir/actions/GestureAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/ChangeHostAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/ChangeProfile.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/Gesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/ReleaseGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/ThresholdGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/IntervalGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/AxisGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/NullGesture.cpp.o src/logid/CMakeFiles/logid.dir/backend/Error.cpp.o src/logid/CMakeFiles/logid.dir/backend/raw/DeviceMonitor.cpp.o src/logid/CMakeFiles/logid.dir/backend/raw/RawDevice.cpp.o src/logid/CMakeFiles/logid.dir/backend/raw/IOMonitor.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/Receiver.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/ReceiverMonitor.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp/Device.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp/Report.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/Error.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/Device.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/Device.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/Error.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/Feature.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/EssentialFeature.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Root.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/FeatureSet.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/DeviceName.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Reset.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/AdjustableDPI.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/SmartShift.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ReprogControls.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/HiresScroll.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ChangeHost.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/WirelessDeviceStatus.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ThumbWheel.cpp.o src/logid/CMakeFiles/logid.dir/util/task.cpp.o src/logid/CMakeFiles/logid.dir/util/ExceptionHandler.cpp.o -o logid  /usr/lib64/libevdev.so  -lconfig++  -ludev  -lgio-2.0  -lgobject-2.0  -lglib-2.0 && :
FAILED: logid 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -O2 -march=x86-64 -pipe -pipe -frecord-gcc-switches -fno-diagnostics-color -fmessage-length=0 -flto -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wall -Wextra -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 src/ipcgull/CMakeFiles/ipcgull.dir/src/function.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/property.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/signal.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/interface.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/node.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/exception.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/common_gdbus.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/server_gdbus.cpp.o src/logid/CMakeFiles/logid.dir/logid.cpp.o src/logid/CMakeFiles/logid.dir/util/log.cpp.o src/logid/CMakeFiles/logid.dir/config/config.cpp.o src/logid/CMakeFiles/logid.dir/InputDevice.cpp.o src/logid/CMakeFiles/logid.dir/DeviceManager.cpp.o src/logid/CMakeFiles/logid.dir/Device.cpp.o src/logid/CMakeFiles/logid.dir/Receiver.cpp.o src/logid/CMakeFiles/logid.dir/Configuration.cpp.o src/logid/CMakeFiles/logid.dir/features/DPI.cpp.o src/logid/CMakeFiles/logid.dir/features/SmartShift.cpp.o src/logid/CMakeFiles/logid.dir/features/HiresScroll.cpp.o src/logid/CMakeFiles/logid.dir/features/RemapButton.cpp.o src/logid/CMakeFiles/logid.dir/features/DeviceStatus.cpp.o src/logid/CMakeFiles/logid.dir/features/ThumbWheel.cpp.o src/logid/CMakeFiles/logid.dir/actions/Action.cpp.o src/logid/CMakeFiles/logid.dir/actions/NullAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/KeypressAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/ToggleHiresScroll.cpp.o src/logid/CMakeFiles/logid.dir/actions/ToggleSmartShift.cpp.o src/logid/CMakeFiles/logid.dir/actions/CycleDPI.cpp.o src/logid/CMakeFiles/logid.dir/actions/ChangeDPI.cpp.o src/logid/CMakeFiles/logid.dir/actions/GestureAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/ChangeHostAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/ChangeProfile.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/Gesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/ReleaseGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/ThresholdGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/IntervalGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/AxisGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/NullGesture.cpp.o src/logid/CMakeFiles/logid.dir/backend/Error.cpp.o src/logid/CMakeFiles/logid.dir/backend/raw/DeviceMonitor.cpp.o src/logid/CMakeFiles/logid.dir/backend/raw/RawDevice.cpp.o src/logid/CMakeFiles/logid.dir/backend/raw/IOMonitor.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/Receiver.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/ReceiverMonitor.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp/Device.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp/Report.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/Error.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/Device.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/Device.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/Error.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/Feature.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/EssentialFeature.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Root.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/FeatureSet.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/DeviceName.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Reset.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/AdjustableDPI.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/SmartShift.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ReprogControls.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/HiresScroll.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ChangeHost.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/WirelessDeviceStatus.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ThumbWheel.cpp.o src/logid/CMakeFiles/logid.dir/util/task. [... cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/std_function.h:334:11: note: type ‘struct function’ itself violates the C++ One Definition Rule
  334 |     class function<_Res(_ArgTypes...)>
      |           ^
lto1: some warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/x86_64-pc-linux-gnu-g++ returned 1 exit status
compilation terminated.
/usr/libexec/gcc/x86_64-pc-linux-gnu/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.```

stvoidit, 2024-06-15:
I don't know if my comment will help in any way, maybe I incorrectly applied some environment variables, but in my case, there were no errors when using clang-18. Unfortunately, I can't try it on the versions below.

my env:
```env
LLVM=-18
CC=clang-18
CXX=clang++-18
LD=lld-18
```

commands:
```bash
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-flto" .. 
make -j16
```

output:
```log
❯ cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-flto" ..
-- The C compiler identification is Clang 18.1.8
-- The CXX compiler identification is Clang 18.1.8
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang-18 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++-18 - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.34.1") 
LogiOps Version Number: v0.3.4
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'gio-2.0'
--   Found gio-2.0, version 2.72.4
-- Checking for module 'glib-2.0'
--   Found glib-2.0, version 2.72.4
--   Build shared library:          OFF
--   Build static library:          ON
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'libevdev'
--   Found libevdev, version 1.12.1
-- Checking for module 'systemd'
--   Found systemd, version 249
-- Checking for module 'libconfig'
--   Found libconfig, version 1.5
-- Checking for module 'libudev'
--   Found libudev, version 249
/home/stvoid/development/github/logiops/src/logid/../ipcgull/src/include
-- systemd units will be installed at /lib/systemd/system
-- dbus system policy will be installed at /usr/share/dbus-1/system.d
-- Configuring done
-- Generating done
-- Build files have been written to: /home/stvoid/development/github/logiops/build
```

Компиляция проходит успешно, logid работает корректно.
[compile_commands.json](https://github.com/user-attachments/files/15855406/compile_commands.json)

ConiKost, 2024-06-15:
@stvoidit And you tested with ```-Werror=odr```?

stvoidit, 2024-06-16:
@ConiKost If this command is correct, then yes, everything was successful. The compilation was completed without errors

```bash
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-flto -Werror=odr" ..
```

ConiKost, 2024-06-17:
I am not able to build. It stills fails for me with ```-march=native -mfpmath=sse,387 -mtune=intel -O3 -frecord-gcc-switches -fomit-frame-pointer -malign-data=abi -mtls-dialect=gnu2 -pipe -flto -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing```. But the difference is, I am using GCC14, not LLVM.

```
FAILED: logid
: && /usr/bin/x86_64-pc-linux-gnu-g++ -march=native -mfpmath=sse,387 -mtune=intel -O3 -frecord-gcc-switches -fomit-frame-pointer -malign-data=abi -mtls-dialect=gnu2 -pipe -flto -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -fvisibility-inlines-hidden -Wall -Wextra -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 -Wl,-O1 -Wl,--sort-common -Wl,-z,now src/ipcgull/CMakeFiles/ipcgull.dir/src/function.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/property.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/signal.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/interface.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/node.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/exception.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/common_gdbus.cpp.o src/ipcgull/CMakeFiles/ipcgull.dir/src/server_gdbus.cpp.o src/logid/CMakeFiles/logid.dir/logid.cpp.o src/logid/CMakeFiles/logid.dir/util/log.cpp.o src/logid/CMakeFiles/logid.dir/config/config.cpp.o src/logid/CMakeFiles/logid.dir/InputDevice.cpp.o src/logid/CMakeFiles/logid.dir/DeviceManager.cpp.o src/logid/CMakeFiles/logid.dir/Device.cpp.o src/logid/CMakeFiles/logid.dir/Receiver.cpp.o src/logid/CMakeFiles/logid.dir/Configuration.cpp.o src/logid/CMakeFiles/logid.dir/features/DPI.cpp.o src/logid/CMakeFiles/logid.dir/features/SmartShift.cpp.o src/logid/CMakeFiles/logid.dir/features/HiresScroll.cpp.o src/logid/CMakeFiles/logid.dir/features/RemapButton.cpp.o src/logid/CMakeFiles/logid.dir/features/DeviceStatus.cpp.o src/logid/CMakeFiles/logid.dir/features/ThumbWheel.cpp.o src/logid/CMakeFiles/logid.dir/actions/Action.cpp.o src/logid/CMakeFiles/logid.dir/actions/NullAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/KeypressAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/ToggleHiresScroll.cpp.o src/logid/CMakeFiles/logid.dir/actions/ToggleSmartShift.cpp.o src/logid/CMakeFiles/logid.dir/actions/CycleDPI.cpp.o src/logid/CMakeFiles/logid.dir/actions/ChangeDPI.cpp.o src/logid/CMakeFiles/logid.dir/actions/GestureAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/ChangeHostAction.cpp.o src/logid/CMakeFiles/logid.dir/actions/ChangeProfile.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/Gesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/ReleaseGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/ThresholdGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/IntervalGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/AxisGesture.cpp.o src/logid/CMakeFiles/logid.dir/actions/gesture/NullGesture.cpp.o src/logid/CMakeFiles/logid.dir/backend/Error.cpp.o src/logid/CMakeFiles/logid.dir/backend/raw/DeviceMonitor.cpp.o src/logid/CMakeFiles/logid.dir/backend/raw/RawDevice.cpp.o src/logid/CMakeFiles/logid.dir/backend/raw/IOMonitor.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/Receiver.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/ReceiverMonitor.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp/Device.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp/Report.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/Error.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp10/Device.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/Device.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/Error.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/Feature.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/EssentialFeature.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Root.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/FeatureSet.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/DeviceName.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Reset.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/AdjustableDPI.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/SmartShift.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ReprogControls.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/HiresScroll.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ChangeHost.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/WirelessDeviceStatus.cpp.o src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ThumbWheel.cpp.o src/logid/CMakeFiles/logid.dir/util/task.cpp.o src/logid/CMakeFiles/logid.dir/util/ExceptionHandler.cpp.o -o logid  /usr/lib64/libevdev.so  -lconfig++  -ludev  -lgio-2.0  -lgobject-2.0  -lglib-2.0 && :
/var/tmp/portage/app-misc/logiops-0.3.4/work/logiops-v0.3.4/src/ipcgull/src/include/ipcgull/variant.h:88:12: error: type ‘struct _y_comb’ violates the C++ One Definition Rule [-Werror=odr]
   88 |     struct _y_comb : K<_y_comb<K>> {
      |            ^
/var/tmp/portage/app-misc/logiops-0.3.4/work/logiops-v0.3.4/src/logid/../ipcgull/src/include/ipcgull/variant.h:88:12: note: a type with different bases is defined in another translation unit
   88 |     struct _y_comb : K<_y_comb<K>> {
      |            ^
/var/tmp/portage/app-misc/logiops-0.3.4/work/logiops-v0.3.4/src/ipcgull/src/include/ipcgull/property.h:80:11: error: type ‘struct base_property’ violates the C++ One Definition Rule [-Werror=odr]
   80 |     class base_property {
      |           ^
[... cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/14/include/g++-v14/bits/std_function.h:334:11: note: type ‘struct function’ itself violates the C++ One Definition Rule
  334 |     class function<_Res(_ArgTypes...)>
      |           ^
lto1: some warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/x86_64-pc-linux-gnu-g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/x86_64-pc-linux-gnu/14/../../../../x86_64-pc-linux-gnu/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

```

stvoidit, 2024-06-18:
To be honest, I don't really understand these tools either. For me, these are just tools, some I trust more, some less.

I don't really understand the essence of the problem, because by default the project is going without full lto? It seems to me that assembling with -flto is saving on matches. I think this is not a project that requires this level of optimization.

I don't know if you need my advice, but using docker to compile static executable files is extremely convenient, or just testing.

offtop: I started using LLVM after in some of my golang projects that use C library dependencies and compiled with GCC 12-14, in some scenarios they gave out a complete crash with creepy traces. After trying to replace GCC with LLVM, I got more stable builds. Since then, I have been compiling fresh releases of the Linux kernel, python, etc. and so on only with clang+llvm. But this is just my experience, maybe for something you need to have a stable compilation with all the tools