PixlOne/logiops #446: MX Master 3S times out

telemann, 2024-03-18:
I can't get logid to automatically recognise my MX  Master 3S. Logid times out looking for it:
`logid -v2` gives me:

`DEBUG] /dev/hidraw3:2 timed out, waiting for input from device to initialize.`

Logid then hangs indefinitely until I press the button on the bottom of the mouse three times to reconnect (or switch it off and on again). Logid then continues:

`[INFO] Device found: MX Master 3S on /dev/hidraw3:2`
`[DEBUG] /dev/hidraw3:2 remappable buttons:`
...

How can I get logid to connect automatically?

telemann, 2024-03-18:
Gentoo Linux

telemann, 2024-03-19:
The timeout in src/logid/Configuration.h is too short. Increasing it from 500 to 1000 (ms?) fixes the problem

````diff -ur a/src/logid/Configuration.h b/src/logid/Configuration.h
--- a/src/logid/Configuration.h 2023-07-12 23:40:47.000000000 +0200
+++ b/src/logid/Configuration.h 2024-03-19 21:19:10.767519123 +0100
@@ -28,7 +28,7 @@

 namespace logid {
     namespace defaults {
-        static constexpr double io_timeout = 500;
+        static constexpr double io_timeout = 1000;
         static constexpr int workers = 4;
         static constexpr int gesture_threshold = 50;
     }

przemekd, 2024-04-10:
I observe the same the same on my system:
```
[INFO] Detected receiver at /dev/hidraw5                                                                                                                                                                                                                                                                                                                                                                                                     
[DEBUG] Unsupported device /dev/hidraw6 ignored                                                                                                                                                                                                                                                                                                                                                                                              
[DEBUG] Unsupported device /dev/hidraw0 ignored                                                                                                                                                                                                                                                                                                                                                                                              
[DEBUG] Unsupported device /dev/hidraw7 ignored                                                                                                                                                                                                                                                                                                                                                                                              
[DEBUG] /dev/hidraw5:2 timed out, waiting for input from device to initialize. 
```
I need to restart `logid` service and the switch off and switch on my mouse to make it working.

BrainFreezy, 2024-04-11:
can confirm that changing the timeout from 500 to 1000 is enough to solve the problem for me

maciejsszmigiero, 2024-05-07:
> The timeout in src/logid/Configuration.h is too short. 
> Increasing it from 500 to 1000 (ms?) fixes the problem

This change fixes the timeout problem with  _MX Master 3S For Business_ for me too.
Are you going to submit this as a PR @telemann ?

telemann, 2024-05-07:
@maciejsszmigiero I figured a developer would find the fix here and merge it. I'm not sure how to do a pull request, if you want to do it please go ahead.

maciejsszmigiero, 2024-05-10:
@PixlOne could you commit such timeout change?

The new timeout could be 5 seconds like [Linux kernel `hid-logitech-hidpp` driver](https://github.com/torvalds/linux/blob/f4345f05c0dfc73c617e66f3b809edb8ddd41075/drivers/hid/hid-logitech-hidpp.c#L310) uses or 4 seconds
like [Solaar](https://github.com/pwr-Solaar/Solaar/blob/bfc06502c124472951a02c3de36116764e87b994/lib/logitech_receiver/base.py#L111) has.

Thanks.

rizaumami, 2024-05-14:
[Isn't it already in the config](https://github.com/PixlOne/logiops/wiki/Configuration#io_timeout)? 
I've set the value to `5000` and `logid` successfully detect MX Master 3S.

```
devices: (
{
    name: "MX Master 3S";
    timeout: 5000;
...
```

telemann, 2024-05-14:
Setting the value in the config doesn't work for me. It looks like the config file is not read until after the mouse has been found. 
Also the documentation is incorrect: the default io_timeout is 500ms, not 2000.

spoutn1k, 2024-05-31:
I had a similar issue and the fix was the patch in #402. I created a PR for this #460, if that fixes your issue please add this issue to the list of related issues.

TomasLiutvinas, 2025-07-14:
Increasing timer like this helped for me. It does not wait full minute, gets registered quickly.
```
io_timeout: 60000.0; # More time to detect the mouse
devices: (
  {
    name: "MX Master 3S";
...
```