PixlOne/logiops #451: Compiler Error: »transform« is not element of »std«

Sebastian0100, 2024-04-29:
During compilation the error »transform« is not element of »std« happens at logiops/src/logid/./config/group.h:151:18. The cause for this seems to be that the header file does not include the Algorithms library, as including it resolves the problem.

Although since nobody else seems to have encountered this issue, this might be related to my specific environment.

Sniptzler, 2024-05-04:
I encountered the same Error but the fix worked seamingless.

debbinkm, 2024-05-05:
I also just ran into this. Adding `#include <algorithm>` to group.h does indeed allow the project to compile, as @Sebastian0100 mentioned. Not sure why this is all of the sudden an issue, though.

kada49, 2024-05-19:
Maybe we all are using a new version of the compiler which changes some includes, no idea

BrainFreezy, 2024-05-19:
thank you that worked for me aswell

PixlOne, 2024-05-28:
Should be fixed with the recent PR.

closed 2024-05-28 by PixlOne