PixlOne/logiops #452: Add SC COPY, PASTE and CUT key

misterbit-pro, 2024-05-02:
Hello,

I like this project a lot because after my switch to linux this is the best sultion to have a Logitech Options like experience like I had on Windows, But I have one problem there are no SC_COPY, CUT and PASTE keys available. Just AC keys wich is unpractical because they do not work in every situation f.e. to copy files from one VS Code project to another. Its just working for text (code) inside the editor. Is there any solution available to add this?

MariusAlbrecht, 2024-05-12:
The usual key combos should work, or am I misunderstanding something? I don't have SC COPY, PASTE and CUT keys on my keyboard either

misterbit-pro, 2024-06-06:
Yeah but I can't copy files and so with the AC keys, maybe there should be an option to select to define key combinations, so it "presses" Ctrl+C for example. That would be nice

misterbit-pro, 2024-06-08:
Oh I see, sorry, I haven't seen that option at first. That fixes it. Thx for your help

closed 2024-06-08 by misterbit-pro