PixlOne/logiops #455: Config for disability zoom in KDE?

b9AcE, 2024-05-09:
Hi!

I'm visually impaired, at hospital-gave-me-a-white-cane level, so ever since I got that I have been relying very heavily on the Windows 10 accessibility fullscreen zoom hotkeys (<win>+<+> to zoom in and <win>+<esc> to end zoom) being bound to the gesture button on my Logitech mice, configured so whenever I need to zoom in, I just hold down the gesture button over whatever needs magnification and move the mouse upwards, then when I am done with being zoomed in hold down the gesture button and move the mouse downwards to end the zooming.
Now Microsoft is making Windows no longer an option and so I must switch back to Linux as desktop but I really, really need to be able to zoom with my Logitech MX Vertical mouse like that.
The keyboard-only native hotkeys for zooming builtin into KDE are insufficient in several ways.

This seems like it should be possible through logiops, right?
I was looking at doing the equivalent of what I did in "Logitech Options" on Windows, but instead in KDE by defining the gestures-button as "<Meta>+<+>" for the up gesture and "<Meta>+0" (zero) for down, but I don't know what those keys are called in the format of logid.cfg, while such a  quantity of text as in the example-config and wiki is practically insurmountable since I got the disability even though I was a UNIX SysAdmin with Linux-focus before that, soo... anyone feel like assisting an officially blind person, me, with this, please?
Thanks!

klinkigt, 2024-05-17:
Hi,

hope this helps:

```
    // Gesture button (hold and move)
    {
      cid: 0xc3;
      action =
      {
        type: "Gestures";
        gestures: (

          {
            direction: "Up";
            mode: "OnInterval";
            interval: 20;
            action =
            {
                type: "Keypress";
                keys: ["KEY_LEFTMETA", "KEY_PLUS"];
            };
          },
          {
            direction: "Down";
            mode: "OnInterval";
            interval: 20;
            action =
            {
                type: "Keypress";
                keys: ["KEY_LEFTMETA", "KEY_0"];
            };
          },
        );
      };
    },
```

Those are only the part you have asked for mapping for META+"+" and META+0 to the gesture key for up and down movement. If you have already a section for the gesture key, only add the inner part.

May share your config here, than I can have a look.

Best

b9AcE, 2024-05-17:
Hey!
Unfortunately I gave up on this already and without a usable zoom functionality I was forced to temporarily go back to Windows while hoping the situation KDE situation improves before Windows 10's support ends,
so I will not be able to test the configuration that you so kindly provided for me now @klinkigt, but I will save it and test it the next time I give the transition an attempt.
Thank you very, very much!

closed 2024-05-17 by b9AcE

b9AcE, 2025-09-28:
Just in case anyone else finds this issue while trying to do the same or something similar, here is what my /etc/logid.cfg ended up containing when working as intended.

It sets the MX Vertical mouse's gestures-capable middle-button to send `<meta>+<numpad +>` on `<gesture>+<mouse up>` and `<meta>+<numpad 0>` on `<gesture>+<mouse down>`, which matches the keyboard shortcuts configured in KDE on this system for "Zoom In and "Zoom to Actual Size" (=end zoom) to easily be able to zoom in exactly where the mouse cursor is while doing the gesture.
A threshold of 100 was set for the zoom-gestures because making a left/right-gesture could accidentally also trigger the zoom in/end at the default 50.

Added is also configuration to switch virtual desktop by mouse gesture left/right as well as finally also the top right button on the MX Keys keyboard is set to do nothing because by default the very clever designer at Logitech made it so that if you accidentally slip across that button e.g. while just intending to lift the keyboard with your hand then it locks the screen, which can have very negative results including loss of work.

```
devices: (
    {
        name: "MX Vertical Advanced Ergonomic Mouse";
        buttons: (
            {
                cid: 0x0052;
                action =
                {
                    type: "Gestures";
                    gestures: (
                        {
                            direction: "Up";
                            mode: "OnThreshold";
                            threshold: 100;
                            action =
                            {
                                type: "Keypress";
                                keys: ["KEY_LEFTMETA", "KEY_KPPLUS"];
                            };
                        },
                        {
                            direction: "Down";
                            mode: "OnThreshold";
                            threshold: 100;
                            action =
                            {
                                type: "Keypress";
                                keys: ["KEY_LEFTMETA", "KEY_KP0"];
                            };
                        },
                        {
                            direction: "Right";
                            mode: "OnThreshold";
                            action =
                            {
                                type: "Keypress";
                                keys: ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_RIGHT"];
                            };
                        },
                        {
                            direction: "Left";
                            mode: "OnThreshold";
                            action =
                            {
                                type: "Keypress";
                                keys: ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_LEFT"];
                            };
                        },
                    );
                };
            },
        );
    },
    {
        name: "MX Keys Wireless Keyboard";
        buttons: (
            {
                cid: 0x006f;
                action =
                {
                    type: "None";
                };
            },
        );
    },
);
```