PixlOne/logiops #463: MX Master 3S with Bolt not detected

kreare, 2024-06-14:
I have an MX Master 3S connected with a Bolt receiver.
Logiopt doesn't detect it on startup. The only way is to start/restart logid when the mouse is moving. If the mouse is moving when logid starts, then it's detected immediatly.

Any idea hwo to fix this kind of issue ?

stvoidit, 2024-06-15:
Just the other day, I upgraded to this mouse. Before that, I used the logid for the "Wireless Mobile Mouse MX Anywhere 2S" mouse.

At the first restart of logid with a new config, the mouse was detected immediately, picked up the configurations, however, while debugging the config in the debug mode `logid -v -c /etc/logid.cfg`, the behavior was strange. Sometimes logid immediately detects the mouse and the configuration is applied, sometimes not. But it works 100% if you turn off and turn on the mouse hardware, the switch from the bottom of the mouse - logid immediately picks it up.

I will give my configuration below. I don't know if it could have had an impact, but now I'm seeing more stable work by setting the `workers=16 parameter`. In general, I would like a clearer explanation about the number of workers and their influence, because documentation and help are extremely scarce, and finding out from sources is not the most interesting thing for me right now.

```conf
workers: 16;
devices: (
{
    name: "MX Master 3S";
    smartshift:
    {
        on: true;
        threshold: 40;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    thumbwheel: {
        divert: true;
        invert: false;
        left: {
            threshold: 50;
            interval: 1;
            direction: "Left";
            mode: "OnInterval";
            action =
            {
                type: "Keypress";
                keys: ["KEY_VOLUMEDOWN"];
            };
        };
        right: {
            threshold: 40;
            interval: 1;
            direction: "Right";
            mode: "OnInterval";
            action =
            {
                type: "Keypress";
                keys: ["KEY_VOLUMEUP"];
            };
        };
    };
    dpi: 4500;
    buttons: (
        {
            cid: 0x56;
            action =
            {
                type: "Keypress";
                keys: ["KEY_PAGEUP"];
            };
        },
        {
            cid: 0x53;
            action =
            {
                type: "Keypress";
                keys: ["KEY_PAGEDOWN"];
            };
        }
    );
}
);
```

kreare, 2024-06-18:
timeout did nothin in my configuraton

the only way to make this detecting the MX master is moving the mouse while starting up logiopt

stvoidit, 2024-06-18:
@kreare yes, my mx 3s is about a week old and I have the same problem. Not that this is critical, but it is quite unpleasant when, when turning on the laptop after logging in, I have to either nervously pull the mouse trying to get into the daemon's timing, or turn off and turn on the mouse power :)

Actually, it seems to me that the problem is with the new receiver. Now I have switched to using bluetooth only and the mouse operation seems to be much more stable.
In addition, due to the specific architecture of the laptop case, I can only insert the receiver from behind and during games, in fact, it is in the hottest zone. After some time of heating, the mouse stops stably catching the signal, the movements appear jerky. 
To be fair, I have such a problem with all receivers because of my laptop :)

stvoidit, 2024-06-19:
@kreare yes, you need to set the `hiresscroll.hires` flag to `true`, then the wheel will work as before.
This is probably the most inconvenient thing, if the creators manage to fix it, then it will be just fine.
Now I have this problem with the wheel on the last logitech mouse.

kreare, 2024-06-21:
with hires set to true is much better but i'm still feeling some lag. i don't think bluetooth is totally comparable to Bolt

fureunoir, 2024-09-09:
> hello, just installed logiops just a few hours ago, and I still couldn't make it work. I'm also have a Mx master 3s, I'm currently using ubuntu 22 and my pc doesn't have bluetooth so using the bolt receiver is my only option.
> 
> here's my logid configuration below. every time i tried to run `sudo logid -c /etc/logid.cfg`, that's my output `[INFO] Detected receiver at /dev/hidraw8` and none of my configuration works... how did u made it work with the bolt receiver
> 
> ` devices: ( { name: "MX Master 3S"; timeout: 1500; smartshift: { on: true; threshold: 30; torque: 50; }; hiresscroll: { hires: true; invert: false; target: false; }; dpi: 1000;
> 
> ```
>     buttons: (
>         {
>             cid: 0xc3;
>             action =
>             {
>                 type: "Gestures";
>                 gestures: (
>                     {
>                         direction: "Up";
>                         mode: "OnRelease";
>                         action =
>                         {
>                             type: "Keypress";
>                             keys: ["KEY_UP"];
>                         };
>                     },
>                     {
>                         direction: "Down";
>                         mode: "OnRelease";
>                         action =
>                         {
>                             type: "Keypress";
>                             keys: ["KEY_DOWN"];
>                         };
>                     },
>                     {
>                         direction: "Left";
>                         mode: "OnRelease";
>                         action =
>                         {
>                             type: "Keypress";
>                             keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_LEFT"];
>                         };
>                     },
>                     {
>                         direction: "Right";
>                         mode: "OnRelease";
>                         action =
>                         {
>                             type: "Keypress";
>                             keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_RIGHT"];
>                         };
>                     },
>                     {
>                         direction: "None";
>                         mode: "OnRelease";
>                         action =
>                         {
>                             type: "Keypress";
>                             keys: ["KEY_LEFTMETA"];
>                         };
>                     }
>                 );
>             };
>         },
>         {
>             cid: 0xc4;
>             action =
>             {
>                 type = "ToggleSmartShift";
>             };
>         }
>     );
> }
> );
> ```
> 
> `

I'm just the same here...

agluck91, 2025-03-14:
I would like to ping this thread again. I am having the same issue with the bolt receiver. It detects the reciever with no troubles, but none of the config values are being applied. I am using the latest version of logiops and I am on the latest unstable of Nixos as well.

TharinduLakshanKumara, 2025-06-15:
> I would like to ping this thread again. I am having the same issue with the bolt receiver. It detects the reciever with no troubles, but none of the config values are being applied. I am using the latest version of logiops and I am on the latest unstable of Nixos as well.

When mouse is connected via Logi Bolt, Ubuntu doesn't identify as a new device. I tried and tried but there is no way Ubuntu identifying it as 'MX Master 3S'. You can only get that configuration work when the mouse is connected via bluetooth. When connected via bluetooth OS identifies the mouse without a problem and 'sudo logid' and 'sudo logid -v' printout the device details. I think Logi Bolt doesn't pass the device information to Ubuntu as we are expecting.

holantomas, 2025-06-16:
Read what I wrote. Just move with your mouse while you want use `logid` :)