PixlOne/logiops #465: M720 freezes 30 sec after returning from suspend when in bluetooth mode

iant321, 2024-06-21:
Problem description: M720 is connected via bluetooth. The bluetooth setting is to switch it on after resume from sleep. PowerManager anf KillSwitch are turned off via blueman-manager.  Now, close the laptop lid, system goes to sleep (RAM), open the lid, my M720 gets connected and works for ~30 sec.  Then it just stops reacting to movement and button clicks. Checking with bluetoothctl shows the status as connected. No any error messages in logs. So it seems that OS thinks the mouse is OK. The recipe to make it working is either switch the mouse off and on physically, or just press the channel change button on the mouse to cycle through  the channels 1,2,3 to the original one. In both cases, the mouse gets connected again and works until the next sleep event. This seems like a minor problem, but still quite annoying when  happens several times every day. I do not want to use my USB receiver as my laptop has only 2 USB-A ports which I would like to keep free. Also, the receiver is quite protruding.

My current OS: Fedora 40 (but it was the same since 38)
Kernel: 6.9.4
DE: KDE, plasma 6 (was the same with 5)
Bluez 5.76
logiops 0.3.3, installed with dnf.

The problem is not observed with a Windows computer so it is definitely not a hardware mouse problem. I asked the same question in the Fedora forum, no solution: [my post in Fedora forum](https://forums.fedoraforum.org/showthread.php?332097-Bluetooth-mouse-freezes-after-resume-from-suspend).

JeanneD4RK, 2024-08-02:
Realtek bluetooth chip by any chance ?

iant321, 2024-08-03:
Bus 001 Device 002: ID 8087:0a2b Intel Corp. Bluetooth wireless interface