PixlOne/logiops #466: Won't compile

hatesigningup, 2024-07-03:
Hello, I can't seem to compile Logid.

Fedora 40, build tools installed as per README.md, source file logiops-0.3.4.tar.gz downloaded and decompressed in ~/Downloads/

Following the supplied instructions (....which sadly **does not** specify from which folder I need to run them) -
To build this project, run:

```
bash
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release ..
make
```

.....returns:

```
me@fedora:~/Downloads/logiops-0.3.4/build$ cmake -DCMAKE_BUILD_TYPE=Release ..
LogiOps Version Number: null
CMake Error at CMakeLists.txt:59 (add_subdirectory):
  The source directory

    /home/me/Downloads/logiops-0.3.4/src/ipcgull

  does not contain a CMakeLists.txt file.

/home/me/Downloads/logiops-0.3.4/src/logid/../ipcgull/src/include
-- systemd units will be installed at /usr/lib/systemd/system
-- dbus system policy will be installed at /usr/share/dbus-1/system.d
-- Configuring incomplete, errors occurred!
```

That failed, so move into the **src** folder and try from there -

```
me@fedora:~/Downloads/logiops-0.3.4/src/build$ cmake -DCMAKE_BUILD_TYPE=Release ..
CMake Warning:
  Ignoring extra path from command line:

   ".."

CMake Error: The source directory "/home/meDownloads/logiops-0.3.4/src" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```

That also failed, so move into the **logid** folder and try from there-

```
me@fedora:~/Downloads/logiops-0.3.4/src/logid/build$ cmake -DCMAKE_BUILD_TYPE=Release ..
/home/me/Downloads/logiops-0.3.4/src/logid/../ipcgull/src/include
-- systemd units will be installed at /usr/lib/systemd/system
-- dbus system policy will be installed at /usr/share/dbus-1/system.d
-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /home/me/Downloads/logiops-0.3.4/src/logid
```

Finally, some result, next command -

```
me@fedora:~/Downloads/logiops-0.3.4/src/logid/build$ make
make: *** No targets specified and no makefile found.  Stop.
```

The build folder was also completely empty.

What am I doing wrong?  Thanks.

kada49, 2024-07-09:
Hey, you probably downloaded the autogenerated tar archive from github, not the one the author uploaded with the release. The difference is that git submodules are not included in the autogenerated one (ipcgull is a git submodule). So, at least that issue should be fixed by downloading the other tar archive

hatesigningup, 2024-07-10:
Thanks for the reply and info - I'll try it again when I have the time.  Regards.

knuxyl, 2025-03-27:
You need to clone the entire repository, or at least add the ipcgull source. When you clone this repo, do
`git clone --recursive https://github.com/PixlOne/logiops`
Recursive will bring in dependencies