PixlOne/logiops #467: Hires scrolling is extremely fast

Ciel-MC, 2024-07-03:
Device: MX Master 3s For Business
Problem: Scrolling when hires: true is extremely fast

According to `libinput` logs, every event that should to be REL_WHEEL_HI_RES is reported as a REL_WHEEL, and a corresponding * 120 on REL_WHEEL, which is what made the scrolling extremely fast: Hires events are sent as full scrolls, essentially multiplying the scrolling speed by 120x.

I tried to solve it but my lsp is completely dead for some reason and can't find where the keycode even goes wrong, so I'm reporting it

Ciel-MC, 2024-07-03:
`libinput record` Log:
```
  - evdev:
    - [  2, 887034,   2,   8,      -2] # EV_REL / REL_WHEEL                -2
    - [  2, 887034,   2,  11,    -240] # EV_REL / REL_WHEEL_HI_RES       -240
    - [  2, 887034,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +7ms
  # Current time is 21:10:25
  - evdev:
    - [  4, 410056,   2,   8,      -2] # EV_REL / REL_WHEEL                -2
    - [  4, 410056,   2,  11,    -240] # EV_REL / REL_WHEEL_HI_RES       -240
    - [  4, 410056,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +1523ms
  - evdev:
    - [  4, 492055,   2,   8,      -2] # EV_REL / REL_WHEEL                -2
    - [  4, 492055,   2,  11,    -240] # EV_REL / REL_WHEEL_HI_RES       -240
    - [  4, 492055,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +82ms
  - evdev:
    - [  4, 522054,   2,   8,      -3] # EV_REL / REL_WHEEL                -3
    - [  4, 522054,   2,  11,    -360] # EV_REL / REL_WHEEL_HI_RES       -360
    - [  4, 522054,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +30ms
  - evdev:
    - [  4, 530053,   2,   8,      -2] # EV_REL / REL_WHEEL                -2
    - [  4, 530053,   2,  11,    -240] # EV_REL / REL_WHEEL_HI_RES       -240
    - [  4, 530053,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +8ms
  - evdev:
    - [  4, 537054,   2,   8,      -3] # EV_REL / REL_WHEEL                -3
    - [  4, 537054,   2,  11,    -360] # EV_REL / REL_WHEEL_HI_RES       -360
    - [  4, 537054,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +7ms
  - evdev:
    - [  4, 545053,   2,   8,      -4] # EV_REL / REL_WHEEL                -4
    - [  4, 545053,   2,  11,    -480] # EV_REL / REL_WHEEL_HI_RES       -480
    - [  4, 545053,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +8ms
  - evdev:
    - [  4, 552053,   2,   8,      -2] # EV_REL / REL_WHEEL                -2
    - [  4, 552053,   2,  11,    -240] # EV_REL / REL_WHEEL_HI_RES       -240
    - [  4, 552053,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +7ms
  - evdev:
    - [  4, 732056,   2,   8,       2] # EV_REL / REL_WHEEL                 2
    - [  4, 732056,   2,  11,     240] # EV_REL / REL_WHEEL_HI_RES        240
    - [  4, 732056,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +180ms
```
Not too hard to see what is going wrong here

Ciel-MC, 2024-07-05:
I fixed it! The problem was that udev rules did not apply as I was using the bolt receiver instead of the mouse directly, as such the name of the device was not known and didn't have quirks applied to it, adding
```
mouse:*:name:Logitech USB Receiver*:
    MOUSE_WHEEL_CLICK_ANGLE=1
    MOUSE_WHEEL_CLICK_COUNT=360
    MOUSE_WHEEL_CLICK_ANGLE_HORIZONTAL=26
    MOUSE_WHEEL_CLICK_COUNT_HORIZONTAL=14
```
into `/etc/udev/hwdb.d/71-logitech-mice.hwdb` makes it work what looks to be flawlessly.

closed 2024-07-05 by Ciel-MC

MithicSpirit, 2025-08-24:
Does that hwdb still work for you? I've tried using it, and I've made sure it's applying to the device with `udevadm info`, but the scroll speed doesn't seem to change.