PixlOne/logiops #473: CVE-2024-45752: D-Bus service allows configuration by any unprivileged user

hyperair, 2024-09-23:
logiops, in its default configuration, allows any unprivileged user to configure its `logid` daemon via an unrestricted D-Bus service, including setting malicious keyboard macros. This could potentially enable privilege escalation with minimal user interaction required.

The most basic proof-of-concept assigns a shell command to all buttons for connected peripherals. A more crafty attacker could tailor this operation to specific software used on the system, possibly monitoring the process list and mapping malicious macros at exactly the right moment.

References:
- [Security audit in SUSE](https://bugzilla.suse.com/show_bug.cgi?id=1225543#c3)
- [Security bug with proof of concept script](https://bugzilla.suse.com/show_bug.cgi?id=1226598)
- [Proof of concept script](https://bugzilla.suse.com/attachment.cgi?id=875600)
- Debian [DSA](https://security-tracker.debian.org/tracker/CVE-2024-45752) [Bug#1082378](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1082378)

PixlOne, 2024-09-28:
@hyperair Thank you for reporting, could you please confirm that this fixes the issue.

hyperair, 2024-09-30:
Yep, I just verified that the updated dbus policy file prevents the exploit script from running

PR #476 (2024-09-28): Fix CVE-2024-45752
Prevents arbitrary users from accessing d-bus interface. Fixes #473. This change now requires any application using the LogiOps D-Bus interface to run as root.

In the future, a more comprehensive permission system will be added.
commit 9495516e (2024-09-28): Fix CVE-2024-45752

Prevents arbitrary users from accessing d-bus interface. Fixes #473.
This change now requires any application using the LogiOps D-Bus
interface to run as root.
closed 2024-10-04 by hyperair