PixlOne/logiops #474: logiops-0.3.4-alt1.x86_64 не находит устройство после смены хоста или выключения мыши

alekmov, 2024-09-25:
logid -v
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[DEBUG] Failed to add device /dev/hidraw3 on try 1, backing off for 500ms

Подключение восстанавливается только после перезагрузки alt.

sviperz, 2024-10-12:
@alekmov did you resolve the issue? It started surfacing for me after upgrading to Ubuntu 24.10. And restarting the service via `sudo systemctl restart logid` doesn't help.

It only works after system boot up, but once the mouse goes out of sleep or just turned off and turned on again it starts happening.

Sn4tz, 2024-11-11:
Having the same problem. Everything worked fine on Fedora, now that i'm on Kubuntu (24.10) it doesn't work anymore.