PixlOne/logiops #479: Thumbwheel behavior is very weird

gro-david, 2024-11-17:
I have been trying to configure the thumbwheel on my MX Master 3S for Mac for a few hours now. I am trying to emulate my three finger left or right gesture on my touchpad. The same can be achieved using the keys Meta+Left and Meta+Right. I would be happy with a behavior as if I was using OnThreshold, so that one scroll makes the Key combination get pressed exactly once. Right now I cannot seem to dial in the interval really well, since it seems to not only depend on the distance I try to scroll but also the _speed_. This makes this really unprecise.

mukulsharma9118, 2024-11-24:
can you please share your cfg file. I am not able to create one for me. I am also using MX Master 3S for Mac

gro-david, 2024-11-25:
```
───────┬────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
       │ File: /etc/logid.cfg
───────┼────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ devices: ({
   2   │     name: "MX Master 3S For Mac";
   3   │ 
   4   │     dpi: 2000;
   5   │ 
   6   │     smartshift: {
   7   │         on: true;
   8   │         threshold: 30;
   9   │     };
  10   │ 
  11   │     hiresscroll: {
  12   │         hires: true;
  13   │         invert: false;
  14   │         target: false;
  15   │         up: {
  16   │             mode: "Axis";
  17   │             axis: "REL_WHEEL";
  18   │             #axis_multiplier: 1;
  19   │             },
  20   │         down: {
  21   │             mode: "Axis";
  22   │             axis: "REL_WHEEL";
  23   │             #axis_multiplier: -1;
  24   │         },
  25   │     };
  26   │     thumbwheel:
  27   │     {
  28   │         divert: true
  29   │         invert: false
  30   │         left: {
  31   │     mode: "OnInterval";
  32   │     interval: 5;
  33   │     direction: "Left";
  34   │             action =
  35   │                         {
  36   │                             type: "Keypress";
  37   │                             keys: ["KEY_LEFTMETA", "KEY_LEFT"];
  38   │                         };
  39   │         }
  40   │         right: {
  41   │     mode: "OnInterval";
  42   │     interval: 5;
  43   │     direction: "Right";
  44   │             action =
  45   │                         {
  46   │                             type: "Keypress";
  47   │                             keys: ["KEY_LEFTMETA", "KEY_RIGHT"];
  48   │                         };
  49   │         }
  50   │     tap: {
  51   │         type: "Keypress";
  52   │         keys: ["KEY_LEFTALT", "KEY_TAB"];
  53   │     }
  54   │     }
  55   │     buttons: (
  56   │         {
  57   │             cid: 0xc3;
  58   │             action =
  59   │             {
  60   │                 type: "Gestures";
  61   │                 gestures: (
  62   │                     {
  63   │                         direction: "Up";
  64   │                         mode: "OnRelease";
  65   │                         action =
  66   │                         {
  67   │                             type: "Keypress";
  68   │                             keys: ["KEY_LEFTMETA", "KEY_A"];
  69   │                         };
  70   │                     },
  71   │                     {
  72   │                         direction: "Down";
  73   │                         mode: "OnRelease";
  74   │                         action =
  75   │                         {
  76   │                             type: "Keypress";
  77   │                             keys: ["KEY_ESC"];
  78   │                         };
  79   │                     },
  80   │                     {
  81   │                         direction: "Left";
  82   │             mode: "OnRelease";
  83   │                         action =
  84   │                         {
  85   │                             type: "Keypress";
  86   │                             keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_RIGHT"];
  87   │                         };
  88   │                     },
  89   │                     {
  90   │                         direction: "Right";
  91   │             mode: "OnRelease";
  92   │                         action =
  93   │                         {
  94   │                             type: "Keypress";
  95   │                             keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_LEFT"];
  96   │                         };
  97   │                     },
  98   │                     {
  99   │                         direction: "None";
 100   │                         mode: "OnRelease";
 101   │                         action =
 102   │                         {
 103   │                             type: "Keypress";
 104   │                             keys: ["KEY_LEFTMETA"];
 105   │                         }
 106   │                     }
 107   │                 );
 108   │             };
 109   │         }
 110   │     );
```

MHarlock, 2024-12-15:
Hi,
try to decrease the interval value, my cfg has "interval: 2"