PixlOne/logiops #482: Can this run in headless mode? (MX Keys on Debian)

eabase, 2024-12-14:
I can't find any package build for this in Debian. 

As I am looking for a headless (non-GUI) implementation for the `MX Keys` using `Bolt` connection, I would like to be able to run this on CLI. Is this possible? How?

knuxyl, 2025-03-27:
This isn't a GUI program, it basically an input event server. It runs in the background and it's configuration is a file at /etc/logid.cfg. There is no CLI interaction with it but there is also no GUI interaction.

eabase, 2025-11-02:
@knuxyl 

> This isn't a GUI program, ... There is no CLI interaction with it but there is also no GUI interaction.

Exactly, I don't need  GUI, just a way to setup the keyboard (from afresh) on a headless Debian (ARM) device. There seem to have been some success on for the Arch linux in #520 for the MX4. Maybe I can use that.