PixlOne/logiops #483: "Failed to start logid.service: Unit logid.service not found." When Trying To Start

theCCoderr, 2024-12-15:
I tried building and installing this app exactly as written in the github but I try to launch/start it, I get this error: "Failed to start logid.service: Unit logid.service not found."

I have created a logid.cfg file in /etc/ and nothing changed.

I tried pulling the github and building in another directory but got the same error.

## Installation Log:
amrbasbous@Laptop:~$ sudo apt install cmake libevdev-dev libudev-dev libconfig++-dev git
[sudo] password for amrbasbous: 
cmake is already the newest version (3.30.3-1).
libevdev-dev is already the newest version (1.13.2+dfsg-1).
libudev-dev is already the newest version (256.5-2ubuntu3.1).
libconfig++-dev is already the newest version (1.5-0.4build2).
git is already the newest version (1:2.45.2-1ubuntu1).
Summary:
  Upgrading: 0, Installing: 0, Removing: 0, Not Upgrading: 0
amrbasbous@Laptop:~$ git clone https://github.com/PixlOne/logiops.git
Cloning into 'logiops'...
remote: Enumerating objects: 3500, done.
remote: Counting objects: 100% (629/629), done.
remote: Compressing objects: 100% (117/117), done.
remote: Total 3500 (delta 570), reused 529 (delta 512), pack-reused 2871 (from 1)
Receiving objects: 100% (3500/3500), 833.37 KiB | 2.31 MiB/s, done.
Resolving deltas: 100% (2522/2522), done.
amrbasbous@Laptop:~$ cd logiops
amrbasbous@Laptop:~/logiops$ mkdir build
cd build
cmake ..
make
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.45.2")
Submodule 'src/ipcgull' (https://github.com/PixlOne/ipcgull.git) registered for path 'src/ipcgull'
Cloning into '/home/amrbasbous/logiops/src/ipcgull'...
Submodule path 'src/ipcgull': checked out 'cd0f9a8cefb5b2545e163fceb249fdbcbaf666aa'
LogiOps Version Number: v0.3.5
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Checking for module 'gio-2.0'
--   Found gio-2.0, version 2.82.1
-- Checking for module 'glib-2.0'
--   Found glib-2.0, version 2.82.1
--   Build shared library:          OFF
--   Build static library:          ON
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Checking for module 'libevdev'
--   Found libevdev, version 1.13.2
-- Checking for module 'systemd'
--   Package 'systemd', required by 'virtual:world', not found
-- Checking for module 'libconfig'
--   Found libconfig, version 1.5
-- Checking for module 'libudev'
--   Found libudev, version 256
/home/amrbasbous/logiops/src/logid/../ipcgull/src/include
-- dbus system policy will be installed at /usr/share/dbus-1/system.d
-- Configuring done (11.9s)
-- Generating done (0.0s)
-- Build files have been written to: /home/amrbasbous/logiops/build
[  1%] Building CXX object src/ipcgull/CMakeFiles/ipcgull.dir/src/function.cpp.o
[... 179 lines cut ...]
##  "sudo logid -v" Output:
amrbasbous@Laptop:~/logiops/build$  sudo logid -v
[DEBUG] Unsupported device /dev/hidraw7 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[INFO] Detected receiver at /dev/hidraw3
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[INFO] Device found: MX Master 3S on /dev/hidraw3:1
[DEBUG] /dev/hidraw3:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)

mattdillon100, 2025-03-04:
I'm having the same issue. Are you also on Ubuntu 24.10? Did you find a fix?

Wielding, 2025-06-28:
You need to apt install systemd-dev.