PixlOne/logiops #484: Where can I find a configuration file for Logitech MX Keys on Debian?

eabase, 2024-12-19:
I have a Debian 12 installation and trying to connect a European, `MX Keys` via either a BT4.3 or the `Bolt` dongle. However, I have not been able to find a configuration file for the Keyboard. 

**Where can I find a complete configuration file for `MX Keys`?**

ExposedCat, 2024-12-26:
You should check [Configuration](https://github.com/PixlOne/logiops/wiki/Configuration) and [example](https://github.com/PixlOne/logiops/blob/main/logid.example.cfg) from docs

eabase, 2024-12-28:
> You should check [Configuration](https://github.com/PixlOne/logiops/wiki/Configuration) and [example](https://github.com/PixlOne/logiops/blob/main/logid.example.cfg) from docs

You obviously didn't look at it yourself, as it clearly states it is for the MX Mouse!

* https://github.com/PixlOne/logiops/blob/main/logid.example.cfg#L3

ExposedCat, 2024-12-28:
@eabase I sent you link to general docs and an example config which is similar to whatever else device you will use with logiops. But huh, you do you

eabase, 2024-12-28:
@jef
Added the following commit: 

https://github.com/PixlOne/logiops/commit/9064ee3c72328b6f95d085cdfddb888e679dd3fd

However, it was never [**`TESTED`**](https://github.com/PixlOne/logiops/blob/main/TESTED.md) as the page claims.

jef, 2024-12-28:
I no longer have that keyboard, but it worked at the time with my testing.

The righteousness with your comment seems a bit much for a community project though so I would review the code of conduct.

Happy New Years everyone ❤️