PixlOne/logiops #487: Doesn't work after device falls asleep

ExposedCat, 2024-12-26:
This was not the case before, but for some time logiops stopped handling device awake so I have to restart it dozens of times a day. From logs:
```[INFO] /dev/hidraw0:2 fell asleep.```
This is not after PC suspend but after mouse (`Wireless Mouse MX Master 3`) is being idle

diabeetus01, 2025-01-02:
I also have this issue, have only been able to work around it by running `sudo systemctl restart logid` to restart the service, would be nice if it didn't keep happening in the first place though.

ExposedCat, 2025-01-02:
I hate the fact Logitech write on their product boxes they support Linux and then we have to rely on a seemingly reverse-engineered 3rd party partially not working software. 

The bug is super weird: sometimes it works right after idle but then fails

ExposedCat, 2025-01-16:
The issue seems to be an installed and running Solaar. Uninstalling it solved the issue

closed 2025-01-16 by ExposedCat