PixlOne/logiops #488: Fails to build with clang-19

NHOrus, 2024-12-30:
uint is apparently defined in default includes in gcc, but is missing in clang;

https://github.com/PixlOne/logiops/blob/628ab937a25724c1f21e7edf25c8e5aaff82c691/src/logid/config/schema.h#L69

see bug https://bugs.gentoo.org/919053

NHOrus, 2024-12-30:
See also https://github.com/PixlOne/logiops/blob/628ab937a25724c1f21e7edf25c8e5aaff82c691/src/logid/InputDevice.cpp#L35

But those two files are only ones with bare uint

NHOrus, 2024-12-30:
but main error, that's visible after typedefing uint in those two files, is:
```
usr/lib/llvm/19/bin/clang++ -DLOGIOPS_VERSION=\"v0.3.5\" -I/var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/. -I/usr/include/libevdev-1.0 -I/var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/../ipcgull/src/include  -O2 -pipe -Wall -Wextra -std=gnu++20 -MD -MT src/logid/CMakeFiles/logid.dir/logid.cpp.o -MF src/logid/CMakeFiles/logid.dir/logid.cpp.o.d -o src/logid/CMakeFiles/logid.dir/logid.cpp.o -c /var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/logid.cpp
In file included from /var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/logid.cpp:19:
In file included from /var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/./DeviceManager.h:22:
In file included from /var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/./backend/raw/DeviceMonitor.h:22:
In file included from /usr/include/c++/v1/string:647:
In file included from /usr/include/c++/v1/string_view:941:
In file included from /usr/include/c++/v1/algorithm:1842:
In file included from /usr/include/c++/v1/__algorithm/for_each.h:16:
In file included from /usr/include/c++/v1/__ranges/movable_box.h:21:
In file included from /usr/include/c++/v1/optional:1292:
In file included from /usr/include/c++/v1/iterator:712:
In file included from /usr/include/c++/v1/__iterator/common_iterator.h:31:
/usr/include/c++/v1/variant:1459:43: error: no matching function for call to object of type 'std::less<void>'
 1459 |     static_assert(is_convertible<decltype(_Operator{}(std::forward<_T1>(__t1), std::forward<_T2>(__t2))), bool>::value,
      |                                           ^~~~~~~~~~~
```

NHOrus, 2024-12-30:
Correction: After some pain, actual error is:
```
/usr/lib/llvm/19/bin/clang++ -DLOGIOPS_VERSION="v0.3.5" -I/var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/. -I/usr/include/libevdev-1.0 -I/var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/../ipcgull/src/include  -O2 -pipe -Wall -Wextra -std=gnu++20 -MD -MT src/logid/CMakeFiles/logid.dir/logid.cpp.o -MF src/logid/CMakeFiles/logid.dir/logid.cpp.o.d -o src/logid/CMakeFiles/logid.dir/logid.cpp.o -c /var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/logid.cpp
/var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/logid.cpp:120:5: error: use of undeclared identifier 'v0'
  120 | )", LOGIOPS_VERSION, argv[0], default_config);
      |     ^
<command line>:1:25: note: expanded from macro 'LOGIOPS_VERSION'
    1 | #define LOGIOPS_VERSION v0.3.5
      |                         ^
/var/tmp/portage/app-misc/logiops-0.3.5/work/logiops-v0.3.5/src/logid/logid.cpp:123:36: error: use of undeclared identifier 'v0'
  123 |                     printf("%s\n", LOGIOPS_VERSION);
      |                                    ^
<command line>:1:25: note: expanded from macro 'LOGIOPS_VERSION'
    1 | #define LOGIOPS_VERSION v0.3.5
      |                         ^
 ```

Jannik2099, 2024-12-30:
note that this is about libc++ specifically, 0.3.5 builds fine for me with clang & libstdc++