PixlOne/logiops #505: Fails to run latest update

rayzorben, 2025-05-05:
❯ sudo logid -v
logid: symbol lookup error: logid: undefined symbol: _ZN9libconfig6Config9writeFileEPKc

/etc
❯ uname -a
Linux cachygram 6.14.5-3-cachyos #1 SMP PREEMPT_DYNAMIC Mon, 05 May 2025 07:12:16 +0000 x86_64 GNU/Linux

rautamiekka, 2025-05-05:
Careful pasting text into Git without using a code block/tag.

rayzorben, 2025-05-05:
> Careful pasting text into Git without using a code block/tag.

I am not sure what you mean by 'careful'. But yes definitely would look better for formatting. Not sure why this comment was necessary.

Cpn-Neemo, 2025-05-06:
Same here for me,

Running Garuda Linux

`garuda-inxi                                                                                                      127 ✘  at 20:29:49  ▓▒░
System:
Kernel: 6.14.4-zen1-2-zen arch: x86_64 bits: 64 compiler: gcc v: 15.1.1
clocksource: tsc avail: hpet,acpi_pm
parameters: BOOT_IMAGE=/@/boot/vmlinuz-linux-zen
root=UUID=664e6422-fc67-4d43-9743-8ecae0eade51 rw rootflags=subvol=@
quiet resume=UUID=bd4655ec-8542-4b33-876e-eef66975e8ae loglevel=3
acpi_enforce_resources=lax ibt=off
Desktop: KDE Plasma v: 6.3.4 tk: Qt v: N/A info: frameworks v: 6.13.0
wm: kwin_wayland vt: 3 dm: SDDM Distro: Garuda base: Arch Linux
Machine:
Type: Desktop System: Gigabyte product: X470 AORUS GAMING 5 WIFI v: N/A
serial: <superuser required>
Mobo: Gigabyte model: X470 AORUS GAMING 5 WIFI-CF
serial: <superuser required> uuid: <superuser required> UEFI: American
Megatrends LLC. v: F65g date: 03/11/2025
CPU:
Info: model: AMD Ryzen 7 2700X bits: 64 type: MT MCP arch: Zen+ gen: 1+
level: v3 note: check built: 2018-21 process: GF 12nm family: 0x17 (23)
model-id: 8 stepping: 2 microcode: 0x800820E
Topology: cpus: 1x dies: 1 clusters: 1 cores: 8 threads: 16 tpc: 2
smt: enabled cache: L1: 768 KiB desc: d-8x32 KiB; i-8x64 KiB L2: 4 MiB
desc: 8x512 KiB L3: 16 MiB desc: 2x8 MiB
Speed (MHz): avg: 3700 min/max: 2200/3700 boost: enabled scaling:
driver: acpi-cpufreq governor: performance cores: 1: 3700 2: 3700 3: 3700
4: 3700 5: 3700 6: 3700 7: 3700 8: 3700 9: 3700 10: 3700 11: 3700 12: 3700
13: 3700 14: 3700 15: 3700 16: 3700 bogomips: 118400
Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Vulnerabilities: <filter>
Graphics:
Device-1: NVIDIA GP106 [GeForce GTX 1060 6GB] vendor: ASUSTeK driver: nvidia
v: 570.144 alternate: nouveau,nvidia_drm non-free: 550-570.xx+
status: current (as of 2025-04; EOL~2026-12-xx) arch: Pascal code: GP10x
process: TSMC 16nm built: 2016-2021 pcie: gen: 3 speed: 8 GT/s lanes: 16
ports: active: none off: DP-2,HDMI-A-1 empty: DP-1,DVI-D-1,HDMI-A-2
bus-ID: 0a:00.0 chip-ID: 10de:1c03 class-ID: 0300
Display: wayland server: X.org v: 1.21.1.16 with: Xwayland v: 24.1.6
compositor: kwin_wayland driver: X: loaded: nvidia unloaded: modesetting
alternate: fbdev,nouveau,nv,vesa gpu: nvidia,nvidia-nvswitch
d-rect: 5760x2160 display-ID: 0
Monitor-1: DP-2 pos: primary,left model: Samsung LU28R55 serial: <filter>
built: 2023 res: mode: 3840x2160 hz: 60 scale: 160% (1.6) to: 2400x1350
dpi: 154 gamma: 1.2 size: 632x360mm (24.88x14.17") diag: 727mm (28.6")
ratio: 16:9 modes: max: 3840x2160 min: 640x480
Monitor-2: HDMI-A-1 pos: right model: Sony TV serial: <filter> built: 2014
res: mode: 1920x1080 hz: 60 scale: 100% (1) dpi: 55 gamma: 1.2
size: 886x498mm (34.88x19.61") diag: 1016mm (40") ratio: 16:9 modes:
max: 1920x1080 min: 640x480
API: EGL v: 1.5 hw: drv: nvidia nouveau drv: nvidia platforms: device: 0
drv: nvidia device: 1 drv: nouveau device: 2 drv: swrast gbm: drv: nvidia
surfaceless: drv: nvidia wayland: drv: nvidia x11: drv: nvidia
API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: nvidia mesa v: 570.144
glx-v: 1.4 direct-render: yes renderer: NVIDIA GeForce GTX 1060
6GB/PCIe/SSE2 memory: 5.86 GiB display-ID: :1.0
API: Vulkan v: 1.4.309 layers: 8 device: 0 type: discrete-gpu name: NVIDIA
GeForce GTX 1060 6GB driver: nvidia v: 570.144 device-ID: 10de:1c03
surfaces: xcb,xlib,wayland
Info: Tools: api: clinfo, eglinfo, glxinfo, vulkaninfo
de: kscreen-console,kscreen-doctor gpu: nvidia-settings,nvidia-smi
wl: wayland-info x11: xdpyinfo, xprop, xrandr
Audio:
Device-1: NVIDIA GP106 High Definition Audio vendor: ASUSTeK
driver: snd_hda_intel v: kernel pcie: gen: 3 speed: 8 GT/s lanes: 16
bus-ID: 0a:00.1 chip-ID: 10de:10f1 class-ID: 0403
Device-2: Advanced Micro Devices [AMD] Family 17h HD Audio
vendor: Gigabyte driver: snd_hda_intel v: kernel pcie: gen: 3 speed: 8 GT/s
[... 65 lines cut ...]
priority: -2 dev: /dev/nvme1n1p3 maj-min: 259:3
Sensors:
System Temperatures: cpu: 51.9 C mobo: 35.0 C
Fan Speeds (rpm): cpu: 1767 mobo: 708
Power: 12v: 12.17 5v: 5.01 3.3v: 3.39 vbat: 3.06
Info:
Memory: total: 16 GiB available: 15.53 GiB used: 5.39 GiB (34.7%)
Processes: 471 Power: uptime: 13m states: freeze,mem,disk suspend: deep
avail: s2idle wakeups: 0 hibernate: platform avail: shutdown, reboot,
suspend, test_resume image: 6.14 GiB services: org_kde_powerdevil,
power-profiles-daemon, upowerd Init: systemd v: 257 default: graphical
tool: systemctl
Packages: pm: pacman pkgs: 2240 libs: 570 tools: octopi,paru Compilers:
clang: 19.1.7 gcc: 15.1.1 Shell: garuda-inxi default: Zsh v: 5.9
running-in: yakuake inxi: 3.3.38
Garuda (2.7.2-1):
System install date:     2025-01-23
Last full system update: 2025-05-06
Is partially upgraded:   No
Relevant software:       snapper NetworkManager dracut nvidia-dkms
Windows dual boot:       No/Undetected
Failed units:`

`sudo logid -v                                                                                           1 ✘  took 5s   at 20:29:45  ▓▒░
logid: symbol lookup error: logid: undefined symbol: _ZN9libconfig6Config9writeFileEPKc`

murkl, 2025-05-06:
I had the same error and was able to fix it with a rebuild: https://aur.archlinux.org/packages/logiops#comment-1022314

closed 2025-05-06 by rayzorben