PixlOne/logiops #506: Cannot use logid - "undefined symbol: _ZN9libconfig6Config9writeFileEPKc

DominikMendel, 2025-05-22:
If i run `logid` I get an error immediately: 
```
❯ logid
logid: symbol lookup error: logid: undefined symbol: _ZN9libconfig6Config9writeFileEPKc
```

I'm not too sure what has changed. Just after a system update this stoppped working. 

On Arch Linux.

tigh-latte, 2025-05-22:
Hit the same thing, found the solution [here](https://github.com/PixlOne/logiops/issues/505#issuecomment-2854432764).

In short: uninstall `logiops`, remove its entry from `~/.cache/yay` (or whatever AUR helper you use). Reinstall.

DominikMendel, 2025-05-23:
> Hit the same thing, found the solution [here](https://github.com/PixlOne/logiops/issues/505#issuecomment-2854432764).
> 
> In short: uninstall `logiops`, remove its entry from `~/.cache/yay` (or whatever AUR helper you use). Reinstall.

This fixed my problem. Thank you so much! 
Yeah I tried to uninstall and re-install, but looks like you really do need to clear the cache (`yay` in my case).

closed 2025-05-23 by DominikMendel