PixlOne/logiops #507: logid.service fails to start due to libconfig 1.8.1 update

IcEarthlight, 2025-06-01:
After upgrading libconfig from 1.8-1 to 1.8.1-1, logid.service fails to start with the following error:
```
$ sudo systemctl restart logid.service
$ sudo systemctl status logid.service
× logid.service - Logitech Configuration Daemon
     Loaded: loaded (/usr/lib/systemd/system/logid.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Mon 2025-06-02 09:29:15 AEST; 2s ago
   Duration: 16ms
 Invocation: 0bda046ddd4d461ebc3c410879333f72
    Process: 29148 ExecStart=/usr/bin/logid (code=exited, status=127)
   Main PID: 29148 (code=exited, status=127)
   Mem peak: 1.9M
        CPU: 3ms

6月 02 09:29:15 ethltspc systemd[1]: Started Logitech Configuration Daemon.
6月 02 09:29:15 ethltspc logid[29148]: /usr/bin/logid: error while loading shared libraries: libconfig++.so.1>
6月 02 09:29:15 ethltspc systemd[1]: logid.service: Main process exited, code=exited, status=127/n/a
6月 02 09:29:15 ethltspc systemd[1]: logid.service: Failed with result 'exit-code'.
```
Downgrading libconfig back to 1.8-1 resolves the issue.
```
$ paru -U /var/cache/pacman/pkg/libconfig-1.8-1-x86_64.pkg.tar.zst
```

System Information:
- OS: Arch Linux
- LogiOps Version: v0.3.5
- libconfig Version: 1.8.1-1

Steps to Reproduce:
1. Upgrade libconfig to 1.8.1-1
2. Try starting logid.service

Relevant Logs:
```
$ journalctl -xeu logid.service
6月 02 09:30:56 ethltspc logid[38761]: /usr/bin/logid: error while loading shared libraries: libconfig++.so.1>
6月 02 09:30:56 ethltspc systemd[1]: logid.service: Main process exited, code=exited, status=127/n/a
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ An ExecStart= process belonging to unit logid.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 127.
6月 02 09:30:56 ethltspc systemd[1]: logid.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit logid.service has entered the 'failed' state with result 'exit-code'.
```

murkl, 2025-06-02:
Same problem here on Arch.

horzadome, 2025-06-02:
Lucky symlink libconfig++.so.15 >  libconfig++.so.11 fixes the issue for me.
Note that you might have to reconnect the dongle after starting; or at least I had to in order to recognize the mouse.